cesanta/mongoose #620: Digest auth with query string

oficial, 2016-02-13:
As pointed in the comments of #573, apparently there is an issue whith Digest auth when the URI has a query string.
The "response" sent by the browser in the "Authorization" header match the one calculated on the server as long as there is no query strings.
I think the browser calculates the "response" value using the full URI, but when i include the "query_string" in the calculation of the "ha2", the response still does not match.

There is a solution for this ? Or any workaround ?

PS: Mongoose version 6.2

exmex, 2016-02-28:
Just to jump onto this conversation.
It seems like nobody cares about issues because mine is opened for over 3 months.

I haven't found a workaround / solution yet. But mainly because I didn't care. but this is what the RFC2617 (https://www.ietf.org/rfc/rfc2617.txt) specification says to all of this
```
 A client SHOULD assume that all paths at or deeper than the depth of
   the last symbolic element in the path field of the Request-URI also
   are within the protection space specified by the Basic realm value of
   the current challenge. A client MAY preemptively send the
   corresponding Authorization header with requests for resources in
   that space without receipt of another challenge from the server.
```

and

```
 The authenticating server must assure that the resource designated by
   the "uri" directive is the same as the resource specified in the
   Request-Line; if they are not, the server SHOULD return a 400 Bad
   Request error. (Since this may be a symptom of an attack, server
   implementers may want to consider logging such errors.) The purpose
   of duplicating information from the request URL in this field is to
   deal with the possibility that an intermediate proxy may alter the
   client's Request-Line. This altered (but presumably semantically
   equivalent) request would not result in the same digest as that
   calculated by the client
```

rojer, 2016-03-01:
fixed by c18828af7b630767bc73d34e4b9e9c3b2ebcabba

closed 2016-03-01 by rojer