cesanta/mongoose #621: HTTPS not working on one server

erappaport, 2016-02-22:
I have two Windows 2012 servers (one in my test environment and one in a production environment at my client). My web server works properly on the test computer (https) but not on the production computer. When attempting to connect to the production computer, the following message comes up: Turn on TLS 1.0, TLS 1.1, and TLS 1.2 in Advanced settings and try connecting to ... again.

The same software and same PEM file are used on both machines.

My client used a debugging tool on the production server to get this screenshot:

[image]

Please let me know what might be going on. Thanks.

erappaport, 2016-03-09:
My client might be willing to run some debugging on the server. It is important to them to have HTTPS running properly.

cpq, 2016-03-10:
Thanks Ethan!
We will be able to jump on it in a week time.

cpq, 2016-09-27:
We have done multiple changes to HTTPS support recently, do you still observe this?

erappaport, 2016-09-27:
The client decided against HTTPS at this time, so I cannot verify the behavior - Ethan

alashkin, 2016-09-29:
Ok, i'm closing issue, feel free to repoen if you'll face it again

closed 2016-09-29 by alashkin