cesanta/mongoose #622: Http Auth problem with Safari

yanngarras, 2016-02-24:
I have integrate mongoose with Http Authentification with : 
s_http_server_opts.document_root = "/root/web_root";
s_http_server_opts.auth_domain = "test.local";
s_http_server_opts.per_directory_auth_file = ".htpasswd";
s_http_server_opts.global_auth_file = "/root/.htpasswd";
s_http_server_opts.enable_directory_listing = "no";

The /root/.htpasswd contains : 
foo:test.local:ccdc2922329b6dbffb4a87ea9e99f466

The server returns :
HTTP/1.1 401 Unauthorized
WWW-Authenticate: Digest qop="auth", realm="test.local", nonce="1456327007"
Content-Length: 0
which seems to be good.

Authentification is OK with Chrome (foo/bar) but doens't work with Safari 9 or Edge, the server says the credentials are not OK.

Does anyone have the same problem ?

cpq, 2016-02-29:
Hi Yann,
Could you record HTTP headers for all 3 requests please - Chrome (successful), and Safari & Edge (unsuccessful), and paste them here, please?

rojer, 2016-02-29:
Safari and presumably Edge use a longer `cnonce` value that does not fit into the buffer Mongoose allocates for it. Fix incoming.

rojer, 2016-02-29:
fixed by 8f0374569e90fbafae8846c6a48c4a7d38150526, verified to work on Safari and Edge

closed 2016-02-29 by rojer

yanngarras, 2016-03-01:
Hi Rojer,

Thank you very much for your fix, it's work like a charm.

Just a remark, in mg_http_check_digest_auth, cnonce has now a length of 33 while in mg_http_create_digest_auth_header it has a 40 length.

For information, based on your correction I searched on the cnonce topic and I have found that Opera was using a cnonce bigger than 40 digits (https://gnunet.org/bugs/view.php?id=1633) but I did try your correction with Opera and it works.  They certainly reduce their cnonce size and the RFC doesn't mention anything on cnonce's size. 

Regards,

Yann

rojer, 2016-03-01:
yeah, it's an opaque string... i don't particularly like the way we parse them into all these statically sized buffers, but unfortunately values can be quoted so we can't leave them in place and just refernce... 
C and string don't mix very well. sigh.