cesanta/mongoose #623: does mongoose support reverse proxy?

wuxiangbing, 2016-02-24:
in my system, some feature need http reverse proxy server。dose mongoose support it?

cpq, 2016-02-29:
No, mongoose does not have a built-in rev. proxy functionality.
It can be easily implemented though.
Could you elaborate on your use case please?

SamBallantyne, 2016-04-14:
Also looking at this functionality :)

mmikulicic, 2016-04-14:
https://github.com/cesanta/mongoose/tree/master/examples/load_balancer is an example of a load balancing reverse proxy

SamBallantyne, 2016-04-27:
@mmikulicic @cpq is there a simpler example? The load balancer is 80% irrelevant to my use case. (I'm just looking to forward HTTP requests).

mmikulicic, 2016-04-27:
IIRC we don't yet have a smaller example than that.
However I don't think it's 80% irrelevant. A http load balancer is a reverse proxy that doesn't have only one backend where all requests are forwarded to, but picks one randomly for each incoming connection.

Once a connection is established, all the http requests that flow through that connection are proxied as in a non load-balancing use case.

The only other place where the load-balancing aspects comes in is when bad backends are removed on connection errors.

Hope this helps you navigating through the example. I encourage you to play with the example, remove the parts you don't need etc; it's a good way to learn.

rojer, 2016-04-27:
in general we do not target reverse proxy functionality with Mongoose. we target embedded applications and this is not something i'd expect to be critical for that, yet it is complex to implement.
feel free to use the example, but no changes to core Mongoose are planned in relation to this.

closed 2016-04-27 by rojer

mamert, 2017-03-13:
404. Does anyone have a copy of the load balancer code? Or a different example?