cesanta/mongoose #624: MG timers depend on mg_mgr_poll timeout

JohnDearlight, 2016-02-25:
First, I love this library, have used it for years for projects in my company.  I recently started evaluating v6.2 and noticed that the timers do not work as I would like.  I included the example code below.  This is the output that it yields.

`$ ./issue-timer 
1456436960.685026 main() - mgr => [0x7ffe1f7ae620]; nc => [0x233c010]
1456436961.686112 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436962.687162 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436962.687197 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.688244 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436963.688278 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.689327 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436964.689360 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436965.690409 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436965.690442 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436966.691494 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436966.691538 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436967.692587 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436967.692620 ev_handler(0x233c010,6,...) - MG_EV_TIMER
^C
`

This is the output that I want to see, a TIMER event every 1/10sec between POLL event.

`$ ./issue-timer
1456436960.685026 main() - mgr => [0x7ffe1f7ae620]; nc => [0x233c010]
1456436961.686112 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436962.687113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436962.787113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436962.887113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436962.887113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436962.987113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.087113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.087113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.187113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.287113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.387113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.487113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.587113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.687162 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436963.687113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.787113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.887113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.887113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.987113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.087113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.087113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.187113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.287113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.387113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.487113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.587113 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436964.687197 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.688244 ev_handler(0x233c010,0,...) - MG_EV_POLL
1456436963.688278 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.788278 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.888278 ev_handler(0x233c010,6,...) - MG_EV_TIMER
1456436963.988278 ev_handler(0x233c010,6,...) - MG_EV_TIMER
^C
`

This is the main fragment of the example code that I used to generate this test.  I can send the rest if it is necessary.

`static int      CONTINUE_LOOP = 1;
static unsigned MGR_POLL_INTERVAL = 1000; // 10sec                                                                                                                                                          
static double   MGR_TIMER_INTERVAL = 100; // 1sec                                                                                                                                                           

// NOTE - we frequently use LOG.  But that is defined in mongoose.h as a CPP macro.                                                                                                                             
void MY_LOG( const char* fmt, ... );

void ev_handler(struct mg_connection *nc, int ev, void *p);

int main( int argc, char** argv ) {
  struct mg_mgr mgr;
  mg_mgr_init( &mgr, NULL );
  struct mg_connection* nc;
  nc = mg_bind( &mgr, ":2357", ev_handler );
  MY_LOG( "main() - mgr => [%p]; nc => [%p]", &mgr, nc );
  do {
    mg_mgr_poll( &mgr, MGR_POLL_INTERVAL );
    mg_set_timer( nc, MGR_TIMER_INTERVAL );
  } while (CONTINUE_LOOP);
  mg_mgr_free( &mgr );
}
`

At this point, I am only asking for your thoughts about this design.  There must be a reason for having the timers work as one-offs after the mg_mgr_poll times-out.  OTOH, we would find it very useful to be able to use Mongoose's loop to trigger TIMER events between POLL events.

rojer, 2016-02-26:
yes, this is a known problem with the current implementation.
we need to take timers into account when calculating sleep time.
should be relatively easy to do, and we plan to do it.

JohnDearlight, 2016-02-26:
Do you plan to do it soon?  If so then I can wait.  Otherwise, it would be very useful for me to have now.  In which case, if I implemented a mod in my copy, would you be willing to consider it as a patch?

I figure there are two approaches.  A simple array on which the poll function executes a linear scan for the min time.  Or a min-heap in which the poll function would peek at its root.

Both have the complication that sometimes one wants to cancel a timer.  For example, a time-out for a response from a handler; the response arrives (we hope frequently) before the time-out whereupon the timer is cancelled.

rojer, 2016-02-26:
we already scan the list of connections before each select() (not proud of it, but we already pay the cost), so examining timers during that is not too big of a burden.
basically, before each sleep we just decide for how long.
should be really simple. but i am so busy this and next week that there's little hope i will get to it...
try adding it, see if it works.

grodansparadis, 2016-02-26:
Hi,

And you have assigned valid certificates to the VSCP daemon?

Actually the version on the way (in the repository) now have the latest 
communication engine in place. I will write up some step by step 
instructions on how to make SSL work on all channels with it. SSL has 
been working with the old communication engine also but as I wanted to 
do this change I haven't and don't want to put so much effort in the old 
stuff. I expect 1.12 Magnesium to be out in a couple of weeks and then 
SSL will be solid on all interfaces. I promise.

/Ake

rojer, 2016-02-26:
@grodansparadis i strongly suspect this was not intended for this issue :)

grodansparadis, 2016-02-26:
Opps sorry!

renecannao, 2016-02-28:
I am also very interested in this bug/limitation to get fixed.
Thank you

rojer, 2016-03-13:
just committed 69215cf9ebe12fb38e3e28f563143626218ceeb1 which should've fixed it.
please reopen if it doesn't work for you.

closed 2016-03-13 by rojer