cesanta/mongoose #625: Compile 6.2 on Visual Studio 2015 (1900)

grodansparadis, 2016-02-28:
Shouldn’t

#if ( (!(defined(_XOPEN_SOURCE) && _XOPEN_SOURCE >= 700) &&         \
     !(defined(_POSIX_C_SOURCE) && _POSIX_C_SOURCE >= 200809L) &&   \
     !(defined(__DARWIN_C_LEVEL) && __DARWIN_C_LEVEL >= 200809L) && \
     !defined(RTOS_SDK) ) ||                                         \
    (defined(_MSC_VER) && _MSC_VER < 1600 ) /*Visual Studio 2010*/)
#define _MG_PROVIDE_STRNLEN
size_t strnlen(const char *s, size_t maxlen);
#endif

in mongoose.h be

#if ( (!(defined(_XOPEN_SOURCE) && _XOPEN_SOURCE >= 700) &&         \
     !(defined(_POSIX_C_SOURCE) && _POSIX_C_SOURCE >= 200809L) &&   \
     !(defined(__DARWIN_C_LEVEL) && __DARWIN_C_LEVEL >= 200809L) && \
     !defined(RTOS_SDK) ) &&                                         \
    (defined(_MSC_VER) && _MSC_VER < 1600 ) /*Visual Studio 2010*/)
#define _MG_PROVIDE_STRNLEN
size_t strnlen(const char *s, size_t maxlen);
#endif

Anyway this does not work in it' current definition.  Always true on >1600

alashkin, 2016-02-29:
Hi @grodansparadis 
What error do you have? We use MSVC2015 for our internal testing and seems have no compile errors.

grodansparadis, 2016-02-29:
Strange. I have MSVC2015 here and the strlen is definitely redefined (with _MG_PROVIDE_STRNLEN) and available elsewhere to,

grodansparadis, 2016-02-29:
Forgott. Error messages are thus a lot of

1>  controlobject.cpp
1>C:\Users\grodansparadis\development\vscp\src\common\mongoose.h(821): warning C4273: 'strnlen': inconsistent dll linkage
1>  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10150.0\ucrt\string.h(357): note: see previous definition of 'strnlen'

rojer, 2016-05-15:
we now compile on VS 2015 as part of our CI process, i don't think this is an issue anymore.

closed 2016-05-15 by rojer