cesanta/mongoose #628: Parse post form correctly

AlexJoz, 2016-03-08:
Hello.
I need to parse data from small form..
there some inputs and file upload button
```
<input type=text name=v1>
<input type=text name=v2>
<input type=file name=image>
and so on...
``` 
How should i correctly parse those values of input?
I had some experiments with `mg_get_http_var` - for variables it works fine, same with `mg_parse_multipart` - it helps to upload a file.
But they seem not to work together at all...
When i tried to grab some data with parse_multipart this way
`if (strcmp(var_name, "v1") == 0)
        {
            myValue = chunk;
        }`
There are many other things in the chunk except the value of variable (the value is present, but not alone)

What is the most correct way of doing this? 
_______
Thnx<
Alex.

alashkin, 2016-09-29:
Do you try to use `mg_get_http_var` and `mg_parse_multipart` in MG_EV_HTTP_REQUEST handler?
`mg_parse_multipart` function is outdated by now, for parsing multipart messages it is better to use events like MG_EV_HTTP_PART_BEGIN, MG_EV_HTTP_PART_DATA etc

alashkin, 2016-10-03:
Feel free to reponen this issue if it is still actual

closed 2016-10-03 by alashkin