cesanta/mongoose #632: 500 Server Error, when serve in a multibyte-named directory

rufushuang, 2016-03-17:
When mongoose serve in a multibyte-named directory, mongoose can show file list well, and showing the correct multibyte-name file/sub-directory name.

But when navigate to any file in this multibyte-named directory, whether the file is with multibyte character or not, mongoose will return a 500 Server Error.

Some fixed issue in PHPDesktop maybe helpful:

https://github.com/cztomczak/phpdesktop/issues/33

rojer, 2016-03-17:
can you post full URL of the page that gives error? i'm interested in exactly how it is encoded.

rufushuang, 2016-03-17:
Here is what Fiddler Inspectors shows:

First, directory with multibyte-name browse list:

    GET http://84.232.189.18:8080/tooltip%e7%a0%94%e7%a9%b6/qtip2%e7%a0%94%e7%a9%b6/ HTTP/1.1
    Accept: text/html, application/xhtml+xml, */*
    Accept-Language: zh-CN
    User-Agent: Mozilla/5.0 (Windows NT 6.1; Trident/7.0; rv:11.0) like Gecko
    Accept-Encoding: gzip, deflate
    Host: 84.232.189.18:8080
    DNT: 1
    Connection: Keep-Alive

list returns:

    HTTP/1.1 200 OK
    Transfer-Encoding: chunked
    Content-Type: text/html; charset=utf-8
    
    44A
    <html><head><title>Index of /tooltip%e7%a0%94%e7%a9%b6/qtip2%e7%a0%94%e7%a9%b6/</title><script>function srt(tb, col)
    ...

When open a file, and 500 return:

    GET http://84.232.189.18:8080/tooltip%e7%a0%94%e7%a9%b6/qtip2%e7%a0%94%e7%a9%b6/a.html HTTP/1.1
    Accept: text/html, application/xhtml+xml, */*
    Referer: http://84.232.189.18:8080/tooltip%e7%a0%94%e7%a9%b6/qtip2%e7%a0%94%e7%a9%b6/
    Accept-Language: zh-CN
    User-Agent: Mozilla/5.0 (Windows NT 6.1; Trident/7.0; rv:11.0) like Gecko
    Accept-Encoding: gzip, deflate
    Connection: Keep-Alive
    DNT: 1
    Host: 84.232.189.18:8080

returns:

    HTTP/1.1 500 Server Error
    Content-Length: 0

rojer, 2016-03-17:
thanks, this is helpful. i wonder where that 500 is generated... will take a look.
are you using binary of source? can you enable debug?
for binary it would be `-D 4`, for source it would be adding `-DMG_ENABLE_DEBUG`

rufushuang, 2016-03-17:
Using the parameter `-D 4`, it's some more interesting...

If i run the `mongoose-free-6.1.exe` from `cmd.exe` - command line console of my Windows 7, in the directory that names contains multi-byte character, with and without the parameter `-D 4`, all two fail with a 404 error on webroot explore.

when move the `mongoose-free-6.1.exe` to `D:\`, without multi-byte, it's OK to list root again. But same failure on get "D:\中文\foo.txt", some multibyte sub-directory then files.

seems like mongoose run as a double-click, differs between run from command line console, on Windows 7?

the 404 return nothing special found in http inspector, it just fails.

rufushuang, 2016-03-17:
Maybe some config issue?

Run with double-click:

[image]

Run with console command:

[image]

You can see when run with console command, the `C:\中文` becomes something weird

rojer, 2016-03-17:
can you please re-download the mongoose binary? i see it was v6.1 which is quite old,i updated it to 6.3, can you try now?

rufushuang, 2016-03-18:
after using version 6.3, all previous 500 Error now change to 404 error.

and

the console command running differ from double-click start, remains the same.

in the config `http://127.0.0.1/__mg_admin` page, I set `debug` to 4, and setting the error_log_file to a `D:\err.log`. Then restart the mongoose, and confirmed the config was set. But nothing happened, 404 is still 404, with a content-length: 0 body. No error log file created.

alashkin, 2016-11-29:
Fixed in https://github.com/cesanta/mongoose/commit/ea7d69060d4cfd1868add3a55e131f15ce0c31ed

closed 2016-11-29 by alashkin