cesanta/mongoose #633: multithreaded_restful_server sample crashes when closing browser

andrewkirillov, 2016-03-23:
1. Run multithreaded_restful_server sample (Built we MSVC 2013)
2. Use browser to get response (tested with IE and FF)
3. Close browser.
4. Application crashes.

>	mangoose_test.exe!mg_send(mg_connection * nc, const void * buf, int len) Line 2622	C
 	mangoose_test.exe!mg_forward(mg_connection * from, mg_connection * to) Line 3006	C
 	mangoose_test.exe!mg_http_transfer_file_data(mg_connection * nc) Line 4904	C
 	mangoose_test.exe!mg_http_handler(mg_connection * nc, int ev, void * ev_data) Line 5117	C
 	mangoose_test.exe!mg_call(mg_connection * nc, void (mg_connection *, int, void *) * ev_handler, int ev, void * ev_data) Line 2119	C
 	mangoose_test.exe!mg_if_poll(mg_connection * nc, __int64 now) Line 2151	C
 	mangoose_test.exe!mg_mgr_handle_conn(mg_connection * nc, int fd_flags, double now) Line 3498	C
 	mangoose_test.exe!mg_mgr_poll(mg_mgr * mgr, int timeout_ms) Line 3806	C
 	mangoose_test.exe!per_connection_thread_function(void * param) Line 3918	C

jmgregory, 2016-06-14:
I think I may be getting the same error whenever I make a request to my HTTP server:

```
I/DEBUG   ( 2096):     #00  pc 0000dc72  /system/bin/loch-parameter-server (mg_send+11)
I/DEBUG   ( 2096):     #01  pc 0000e12b  /system/bin/loch-parameter-server (mg_forward+10)
I/DEBUG   ( 2096):     #02  pc 0000e26f  /system/bin/loch-parameter-server (mg_http_transfer_file_data+266)
I/DEBUG   ( 2096):     #03  pc 0000e67d  /system/bin/loch-parameter-server (mg_http_handler+112)
I/DEBUG   ( 2096):     #04  pc 0000d117  /system/bin/loch-parameter-server (mg_call+126)
I/DEBUG   ( 2096):     #05  pc 0000d42f  /system/bin/loch-parameter-server (mg_close_conn+42)
I/DEBUG   ( 2096):     #06  pc 0000db8d  /system/bin/loch-parameter-server (mg_mgr_poll+644)
I/DEBUG   ( 2096):     #07  pc 0000dc51  /system/bin/loch-parameter-server (per_connection_thread_function+28)
I/DEBUG   ( 2096):     #08  pc 0000ca78  /system/lib/libc.so (__thread_entry+72)
I/DEBUG   ( 2096):     #09  pc 0000cbf4  /system/lib/libc.so (pthread_create+208)
```

mg_send+11 points to the first `if` statement here:

```
void mg_send(struct mg_connection *nc, const void *buf, int len) {
  nc->last_io_time = mg_time();
  if (nc->flags & MG_F_UDP) {                                                                                                            
    mg_if_udp_send(nc, buf, len);
  } else {
    mg_if_tcp_send(nc, buf, len);
  }
#if !defined(NO_LIBC) && !defined(MG_DISABLE_HEXDUMP)
  if (nc->mgr && nc->mgr->hexdump_file != NULL) {
    mg_hexdump_connection(nc, nc->mgr->hexdump_file, buf, len, MG_EV_SEND);
  }
#endif
}
```

What's odd is that from the client side (curl) the request seems to complete just fine and return the appropriate response.

jmgregory, 2016-06-14:
I should clarify that the error I'm getting is a SEGFAULT.

jmgregory, 2016-06-14:
Found I could work around this by adding `-DMG_DISABLE_CGI` to my compiler flags.  (Didn't care about CGI anyways.)  Actually it might be better to add `-DMG_DISABLE_FILESYSTEM` instead, assuming you don't need that feature.

alashkin, 2016-08-16:
Should be fixed in 2efc859

closed 2016-08-16 by alashkin