cesanta/mongoose #635: new "endpoints" issue

magnusmanske, 2016-03-26:
Updated to latest version. Now can't compile:

```
mongoose.c: In function ‘mg_destroy_conn’:
mongoose.c:2160:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
   mbuf_free(&conn->endpoints);
                  ^
mongoose.c: At top level:
mongoose.c:3042:13: error: static declaration of ‘mg_sock_set’ follows non-static declaration
 static void mg_sock_set(struct mg_connection *nc, sock_t sock);
             ^
In file included from mongoose.c:1:0:
mongoose.h:1575:6: note: previous declaration of ‘mg_sock_set’ was here
 void mg_sock_set(struct mg_connection *nc, sock_t sock);
      ^
mongoose.c: In function ‘mg_http_get_endpoint_handler’:
mongoose.c:4927:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
   while (pos < nc->endpoints.len) {
                  ^
mongoose.c:4929:25: error: ‘struct mg_connection’ has no member named ‘endpoints’
     memcpy(&name_len, nc->endpoints.buf + pos, sizeof(name_len));
                         ^
mongoose.c:4930:40: error: ‘struct mg_connection’ has no member named ‘endpoints’
     if ((matched = mg_match_prefix_n(nc->endpoints.buf + pos + sizeof(size_t),
                                        ^
mongoose.c:4936:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
                nc->endpoints.buf + pos + sizeof(name_len) + (name_len + 1),
                  ^
mongoose.c: In function ‘mg_register_http_endpoint’:
mongoose.c:7477:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
   mbuf_append(&nc->endpoints, &len, sizeof(len));
                  ^
mongoose.c:7478:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
   mbuf_append(&nc->endpoints, uri_path, len + 1);
                  ^
mongoose.c:7479:18: error: ‘struct mg_connection’ has no member named ‘endpoints’
   mbuf_append(&nc->endpoints, &handler, sizeof(handler));
                  ^
```

rojer, 2016-03-26:
looks like you have not updated mongoose.h

cpq, 2016-03-28:
@magnusmanske please reopen if still the issue.

closed 2016-03-28 by cpq