cesanta/mongoose #636: mongoose.h - LOG() macro

JohnDearlight, 2016-04-04:
The LOG cpp macro conflicts the identical identifier for a log function that we use all over our code, we have for years.

For years also, we usually simply comment it in mongoose.h when we import a new version.  Previous versions of Mongoose did not use it, not in mongoose.h (to define it) nor in mongoose.c, making this a mindlessly easy task.

However, recent version of mongoose.c do use the LOG macro extensively, making it a non-trivial task for us to eliminate the conflict.

We respectfully ask that the name of this macro be changed to conform to your general name scheme in the rest of your code in which you prefix your identifiers with "mg_".   Perhaps MG_LOG is acceptable as it conforms to your name scheme and should be minimally disruptive.

We would not ask this except that
- LOG is a very generic ID that many people use.
- Being a library, Mongoose should strive to minimize potential conflicts of all kinds except where unavoidable

We hope that we are not asking too much.

rojer, 2016-04-05:
fair point about LOG being too generic, i'll rename it.

alashkin, 2016-08-23:
I removed `LOG` from headers, so, it should not be a problem 
You can undefine _your_ LOG & friends for mongoose.c

Feel free to reopen issue if I misunderstand problem

closed 2016-08-23 by alashkin