cesanta/mongoose #640: Using Mongoose with VisualGDB project templates for ESP8266

benjamind, 2016-04-10:
I'm attempting to add Mongoose to my VisualGDB project based off of the sample LEDBlink template using the RTOS SDK for ESP8266.

I added the mongoose.h and .c file to the project, copied in the esp_libc.c from the RTOS_SDK example and added the preprocessor defines from the Makefile in the RTOS_SDK user folder. No matter what combination of settings I use I end up with undefined references to various stdio functions (vfprintf, fputc, fflush, __asert_func, strerror etc). Full log is below.

Any idea whats going on here? It would be really great to get a VisualGDB project template setup with Mongoose (and v7).

```bash
1>------ Build started: Project: Mongoose2, Configuration: Debug Win32 ------
1>  VisualGDB: Run "cmd.exe /c "C:\SysGCC\esp8266\bin\make.exe" CONFIG=Debug -j4" in directory "E:\Workspace\VisualGDB\Mongoose2\Mongoose2" on local computer
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-gcc.exe -ggdb -ffunction-sections -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/gpio.c -o Debug/gpio.o -MD -MF Debug/gpio.dep
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-gcc.exe -ggdb -ffunction-sections -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/hw_timer.c -o Debug/hw_timer.o -MD -MF Debug/hw_timer.dep
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-gcc.exe -ggdb -ffunction-sections -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/uart.c -o Debug/uart.o -MD -MF Debug/uart.dep
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-gcc.exe -ggdb -ffunction-sections -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c C:\SysGCC\esp8266\esp8266-bsp/sysprogs/stubs.c -o Debug/stubs.o -MD -MF Debug/stubs.dep
1>  C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/hw_timer.c: In function 'hw_timer_init':
1>C:\SysGCC\esp8266\esp8266-bsp\RTOS-SDK\examples\driver_lib\driver\hw_timer.c(79,41): warning : passing argument 2 of '_xt_isr_attach' from incompatible pointer type [-Wincompatible-pointer-types]
1>       _xt_isr_attach(ETS_FRC_TIMER1_INUM, hw_timer_isr_cb, NULL);
1>                                           ^
1>  In file included from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/freertos/portable.h:318:0,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/freertos/FreeRTOS.h:87,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/spiffs/spiffs_config.h:18,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/spiffs/spiffs.h:16,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp_spiffs.h:28,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp_common.h:102,
1>                   from C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/hw_timer.c:25:
1>  C:\SysGCC\esp8266\esp8266-bsp\RTOS-SDK\include\freertos\portmacro.h(186,13): note:  expected '_xt_isr {aka void (*)(void *)}' but argument is of type 'void (*)(void)'
1>   void        _xt_isr_attach          (uint8 i, _xt_isr func, void *arg);
1>               ^
1>  C:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/driver/uart.c: In function 'UART_SetBaudrate':
1>C:\SysGCC\esp8266\esp8266-bsp\RTOS-SDK\examples\driver_lib\driver\uart.c(238,5): warning : implicit declaration of function 'uart_div_modify' [-Wimplicit-function-declaration]
1>       uart_div_modify(uart_no, UART_CLK_FREQ / baud_rate);
1>       ^
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-gcc.exe -ggdb -ffunction-sections -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c esp_libc.c -o Debug/esp_libc.o -MD -MF Debug/esp_libc.dep
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-g++.exe -ggdb -ffunction-sections -fno-exceptions -fno-rtti -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM=1 -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVA [... cut ...]
1>  Directory: E:\Workspace\VisualGDB\Mongoose2\Mongoose2
1>VisualGDB : error : Command-line action failed
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(37,5): error MSB3073: The command ""C:\Program Files (x86)\Sysprogs\VisualGDB\\VisualGDB.exe" /build "E:\Workspace\VisualGDB\Mongoose2\Mongoose2\Mongoose2.vcxproj" "/solution:E:\Workspace\VisualGDB\Mongoose2\Mongoose2.sln" "/config:Debug" "/platform:Win32"" exited with code 1.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

rojer, 2016-04-10:
the hw_timer error seems unrelated to mongoose - we don't have it in the example, where does this code come from?

as to linker error, it seems that libc is not being linked in, even though headers are present at the time of compilation.
whatever `libminic` is (inferring from `-lminic` on the command line), it doesn't have I/O functions.
if you don't need filesystem i/o, try building with `-DMG_DISABLE_FILESYSTEM` and further, to tirn off logging to stderr, `-DMG_DISABLE_STDIO`.

benjamind, 2016-04-10:
The hw_timer error is from a set of ESP8266 peripheral drivers for the RTOS SDK that VisualGDB includes with the project. I'm happy to live with the warnings caused by these as they provide convenience gpio and uart methods. I've removed them before along with the code that uses them and I still get the errors compiling mongoose.

I've tried building with MG_DISABLE_FILESYSTEM and MG_DISABLE_STDIO and now I get undefined references to __assert_func, gettimeofday, strtoll, memchr and time. I've attached the log below, and here's a copy of the main cpp file http://pastebin.com/4qZKCXp1

```bash
1>------ Build started: Project: Mongoose2, Configuration: Debug Win32 ------
1>  VisualGDB: Run "cmd.exe /c "C:\SysGCC\esp8266\bin\make.exe" CONFIG=Debug -j4" in directory "E:\Workspace\VisualGDB\Mongoose2\Mongoose2" on local computer
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-g++.exe -ggdb -ffunction-sections -fno-exceptions -fno-rtti -O0   -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals -I. -IC:\SysGCC\esp8266\esp8266-bsp/sysprogs/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/espressif/esp8266 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv4 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include/lwip/ipv6 -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/extra_include -IC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/examples/driver_lib/include -DDEBUG=1 -DCS_PLATFORM=3 -DMG_NO_BSD_SOCKETS -DMG_DISABLE_FILESYSTEM -DRTOS_SDK -DMG_LWIP -DLWIP_TIMEVAL_PRIVATE=0 -DMG_ESP8266 -DMG_DISABLE_STDIO -DCS_DISABLE_STDIO -DMG_INTERNAL= -DICACHE_FLASH -DGDBSTUB_BREAK_ON_INIT=1 -DGDBSTUB_FREERTOS=1  -c LEDBlink.cpp -o Debug/LEDBlink.o -MD -MF Debug/LEDBlink.dep
1>  C:\SysGCC\esp8266/bin/xtensa-lx106-elf-g++.exe -o Debug/Mongoose2.elf -Wl,-gc-sections -lmain -TC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/ld/eagle.app.v6.ld  -nostdlib -mlongcalls -mtext-section-literals -nostdlib -mlongcalls -mtext-section-literals  -LC:\SysGCC\esp8266/hal -LC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/lib -LC:\SysGCC\esp8266\esp8266-bsp/RTOS-SDK/ld -Wl,--start-group Debug/gpio.o Debug/hw_timer.o Debug/uart.o Debug/stubs.o Debug/esp_libc.o Debug/LEDBlink.o Debug/mongoose.o -lminic -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lfreertos -llwip -lcrypto -lwpa    -Wl,--end-group
1>Debug/mongoose.o : error VGDB1002: undefined reference to `gettimeofday'
1>Debug/mongoose.o : error VGDB1002: undefined reference to `gettimeofday'
1>Debug/mongoose.o : error VGDB1002: undefined reference to `__assert_func'
1>  Debug/mongoose.o: In function `mbuf_insert':
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `__assert_func'
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `__assert_func'
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `__assert_func'
1>Debug/mongoose.o : error VGDB1002: undefined reference to `__assert_func'
1>  Debug/mongoose.o:E:\Workspace\VisualGDB\Mongoose2\Mongoose2/mongoose.c:9304: more undefined references to `__assert_func' follow
1>Debug/mongoose.o : error VGDB1002: undefined reference to `strtoll'
1>  Debug/mongoose.o: In function `mg_http_parse_headers':
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `strtoll'
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `memchr'
1>  Debug/mongoose.o: In function `mg_parse_http':
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `memchr'
1>  Debug/mongoose.o: In function `mg_handle_chunked':
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `__assert_func'
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `__assert_func'
1>  Debug/mongoose.o: In function `mg_match_prefix_n':
1>E:\Workspace\VisualGDB\Mongoose2\Mongoose2\mongoose.c(9304): error VGDB1000: undefined reference to `memchr'
1>Debug/mongoose.o : error VGDB1002: undefined reference to `time'
1>Debug/mongoose.o : error VGDB1002: undefined reference to `time'
1>collect2.exe : error : ld returned 1 exit status
1>  make: *** [Debug/Mongoose2.elf] Error 1
1>  -------------------------------------------------------------
1>  Command exited with code 2
1>  Executable: cmd.exe
1>  Arguments: /c "C:\SysGCC\esp8266\bin\make.exe" CONFIG=Debug -j4
1>  Directory: E:\Workspace\VisualGDB\Mongoose2\Mongoose2
1>VisualGDB : error : Command-line action failed
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(37,5): error MSB3073: The command ""C:\Program Files (x86)\Sysprogs\VisualGDB\\VisualGDB.exe" /build "E:\Workspace\VisualGDB\Mongoose2\Mongoose2\Mongoose2.vcxproj" "/solution:E:\Workspace\VisualGDB\Mongoose2\Mongoose2.sln" "/config:Debug" "/platform:Win32"" exited with code 1.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```

benjamind, 2016-04-10:
FYI, libminic is provided with the RTOS SDK for esp8266:

https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/lib

rojer, 2016-04-10:
RTOS SDK also comes with full-fledged libc based on newlib. it is then objcopied to rom segment to produce `libcirom.a`
can you try compiling with `-lcirom` instead?

benjamind, 2016-04-10:
Yep, just figured this one out myself...

I added this PR (https://github.com/cesanta/mongoose/pull/641) to give you a copy of what I've been working with. This is tested under the latest VisualGDB as working including jtag debugging.

benjamind, 2016-04-10:
Only thing remaining to figure out is an eagle.app.v6.ld configuration to setup a 1M code 3M SPIFFS memory configuration. I ended up using your eagle.app.v6.512.compact.ld variant for now as this at least gets it to compile, but I'd like to expand it to the 4096 but don't yet know enough about this to make the changes myself.

alashkin, 2016-09-23:
Added "enhancement" label, because this 

> It would be really great to get a VisualGDB project template setup with Mongoose (and v7).

is actually a feature request (with PR supplied)

cpq, 2020-12-13:
Closing this, as #641 was closed.

closed 2020-12-13 by cpq