cesanta/mongoose #642: Mongoose closes connections on load

marioborges1337, 2016-04-12:
Mongoose seems to be closing connections when the server is on load.
I am using the [restful_server example](https://github.com/cesanta/mongoose/tree/master/examples/restful_server) provided by this repo:

```
$ gcc mongoose.c restful_server.c -o restful_server && ./restful_server
Starting RESTful server on port 8000
```

Then, in another terminal:
```
$ weighttp -n 10000 -c 5000 -t 1 localhost:8000
weighttp 0.3 - a lightweight and simple webserver benchmarking tool

starting benchmark...
spawning thread #1: 5000 concurrent requests, 10000 total requests
progress:  10% done
progress:  20% done
progress:  30% done
progress:  40% done
progress:  50% done
progress:  60% done
progress:  70% done
progress:  80% done
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
[... 478 lines cut ...]
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
error: read() failed: Connection reset by peer (104)
progress: 100% done

finished in 15 sec, 310 millisec and 410 microsec, 653 req/s, 2410 kbyte/s
requests: 10000 total, 10000 started, 10000 done, 9460 succeeded, 540 failed, 0 errored
status codes: 9460 2xx, 0 3xx, 0 4xx, 0 5xx
traffic: 37792700 bytes total, 823020 bytes http, 36969680 bytes data
```

Why does this happen? Please advise.

grodansparadis, 2016-05-19:
I can confirm this to.  Same behaviour over websockets to. At some point websocket events stop coming.  But happens in the "normal" webserver also.

alashkin, 2016-08-24:
I don't think it is about Mongoose.
More likely it is about linux defaults.
Change open files maximum number
ex: `ulimit -n 10000`
And should be ok:
```
weighttp 0.4 - a lightweight and simple webserver benchmarking tool

starting benchmark...
spawning thread #1: 5000 concurrent requests, 10000 total requests
progress:  10% done
progress:  20% done
progress:  30% done
progress:  40% done
progress:  50% done
progress:  60% done
progress:  70% done
progress:  80% done
progress:  90% done
progress: 100% done

finished in 39 sec, 491 millisec and 959 microsec, 253 req/s, 501 kbyte/s
requests: 10000 total, 10000 started, 10000 done, 9323 succeeded, 677 failed, 0 errored
status codes: 9323 2xx, 0 3xx, 0 4xx, 677 5xx
traffic: 20276367 bytes total, 2282977 bytes http, 17993390 bytes data
```
You can find details here: http://stackoverflow.com/a/34645

closed 2016-08-24 by alashkin