cesanta/mongoose #643: mg_serve_http broke in v6.3 and v6.4

erappaport, 2016-04-12:
When i run mongoose 6.2, my site works properly (using HTTP). When I swap out mongoose with either v6.3 or v6.4, none of my static pages are sent to the browser properly.

rojer, 2016-04-12:
i'm afraid i'll need something more concrete than just "not sent properly".
is it sent at all, just not completely or do you get errors?

erappaport, 2016-04-12:
I confirmed that mg_serve_http is being called with the proper uri (and document_root is correct in the options). In v6.2 images, js files, css files all come through properly. In v6.3 and v6.4 those files don't reach the browser somehow. png images, for example, show up as a blank spot.

rojer, 2016-04-12:
ok, let's try to unpick this. enter direct url to one of, say, JS file sin the browser - what happens?
and what OS are you using?

erappaport, 2016-04-12:
[image]
I have attached an image of the problem. This is on Windows 10.

erappaport, 2016-04-12:
The actual png looks like this...
[image]

rojer, 2016-04-12:
this looks like image was not fully served.
ok, next please enable debug log and repeat - run mongoose.exe with `-D 4`
do it from the console (cmd.exe), refresh the page in the browser. there will be a lot of output, please capture it all and attach here.

erappaport, 2016-04-13:
I don't use mongoose.exe - I use mongoose.c and mongoose.h in my own server code. It is a windows program and does not have any console outputs.

rojer, 2016-04-13:
ok. early in the program, please do the following:

FILE *f = fopen("mongoose.log", "w");
cs_log_set_file(f);
cs_log_set_level(4);

erappaport, 2016-04-13:
cs_log_set_file and cs_log_set_level are undefined by the linker. What #define do I have to use? I tried CS_ENABLE_DEBUG, MG_ENABLE_DEBUG and I tried #undef CS_DISABLE_STUDIO

erappaport, 2016-04-13:
With the logging enabled, the png appears when requested alone (I tried 3 times). When I try to load a full page, some of the css and javascript didn't come through. Here's the output:

..\WebCode\ThirdParty\mongoose\mongoose.c:3644 select @ 291794 num_ev=0 of 1, timeout=3000
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CE05E0 fd=848 fd_flags=0 nc_flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CE05E0 proto ev=0 ev_data=04A8FB54 flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CE05E0 user ev=0 ev_data=04A8FB54 flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CE05E0 after user flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CE05E0 after proto flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CE05E0 after fd=848 nc_flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3644 select @ 291797 num_ev=1 of 1, timeout=3000
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CE05E0 fd=848 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 00C7A5E0 00CB57A0
..\WebCode\ThirdParty\mongoose\mongoose.c:2620 00CE05E0 00CB57A0 -1 0, 00000000 00000000
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB57A0 1184
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB57A0 proto ev=1 ev_data=00CB57B8 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB57A0 user ev=1 ev_data=00CB57B8 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB5C20 1192
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 04A8F7C0 00CB5C20
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB5E60 960
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 04A8F7C0 00CB5E60
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 00C7A5E0 00CB5C20
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB57A0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB57A0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3644 select @ 291797 num_ev=1 of 3, timeout=3000
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CB5C20 fd=1192 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5C20 user ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5C20 after user flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CB5C20 after fd=1192 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CB57A0 fd=1184 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB57A0 proto ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB57A0 user ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB57A0 after user flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB57A0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CB57A0 after fd=1184 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CE05E0 fd=848 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 00C7A5E0 00CB5EF0
..\WebCode\ThirdParty\mongoose\mongoose.c:2620 00CE05E0 00CB5EF0 -1 0, 00000000 00000000
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB5EF0 912
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5EF0 proto ev=1 ev_data=00CB5F08 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3549 057EFBAC using select()
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5EF0 user ev=1 ev_data=00CB5F08 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2208 ==================================
..\WebCode\ThirdParty\mongoose\mongoose.c:2209 init mgr=057EFBAC
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 057EFBAC 00CB5E60
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB5830 1196
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 04A8F7C0 00CB5830
..\WebCode\ThirdParty\mongoose\mongoose.c:3544 00CB5D40 1200
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 04A8F7C0 00CB5D40
..\WebCode\ThirdParty\mongoose\mongoose.c:2075 00C7A5E0 00CB5830
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5EF0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5EF0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3644 select @ 291797 num_ev=1 of 5, timeout=3000
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CB5830 fd=1196 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5830 user ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5830 after user flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CB5830 after fd=1196 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CB5EF0 fd=912 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3549 058EFC44 using select()..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5EF0 proto ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0

..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB5EF0 user ev=0 ev_data=04A8FB54 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5EF0 after user flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB5EF0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CB5EF0 after fd=912 nc_flags=0 rmbl=0 smbl=0
[... cut ...]
..\WebCode\ThirdParty\mongoose\mongoose.c:3457 00CB54D0 fd=1264 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB54D0 proto ev=0 ev_data=05AEFBA0 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2102 00CB54D0 user ev=0 ev_data=05AEFBA0 flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB54D0 after user flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:2133 00CB54D0 after proto flags=0 rmbl=0 smbl=0
..\WebCode\ThirdParty\mongoose\mongoose.c:3519 00CB54D0 after fd=1264 nc_flags=0 rmbl=0 smbl=0

rojer, 2016-04-14:
if you can tell which file was not fully served, we can trace its connection.

erappaport, 2016-04-14:
It was probably jquery.min.js

rojer, 2016-04-22:
i think https://github.com/cesanta/mongoose/commit/55398684ccfcbc13b42db114ac248d9a2b4ac762 might have helped, can you try latest mongoose.c?

erappaport, 2016-04-22:
The latest mongoose.c/mongoose.h no longer compiles in Visual Studio:
S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1145) : error C2732: linkage specification contradicts earlier specification for 'opendir'
        S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1145) : see declaration of 'opendir'
S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1146) : error C2732: linkage specification contradicts earlier specification for 'closedir'
        S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1146) : see declaration of 'closedir'
S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1147) : error C2732: linkage specification contradicts earlier specification for 'readdir'
        S:\MMG\MMGPCap\WebCode\ThirdParty\../ThirdParty/mongoose/mongoose.h(1147) : see declaration of 'readdir'
NMAKE : fatal error U1077: '"c:\Program Files (x86)\Microsoft Visual Studio 12.0\vc\bin\cl.exe"' : return code '0x2'

rojer, 2016-04-22:
weird that our CI build didn't break (we also build with MSVC), but https://github.com/cesanta/mongoose/commit/7fdb50807cfe2673ce36bd8c1c7bf66c2edebf95 should've fixed it. please try now.

erappaport, 2016-04-22:
It compiles now but works the same as v6.4 for me. I'll have to continue using v6.2 for now.

jwueller, 2016-05-09:
I am getting the same (or at least similar) symptoms with `mg_serve_http()` as well. MinGW (GCC 4.9.3), 32-bit build on 64-bit Windows. `cURL` reports that the `Content-Length` header is all over the place.

Some further debugging seems to suggest that the problem originates from `mg_stat()`, returning non-sensical values for `st_size`. `cs_stat_t` and `_stati64` do not have the same size (the former uses `_off_t`, while the latter uses `_off64_t`, causing `st_size` to read part of the following timestamp (usually producing quite large numbers, obviously). #644 seems to be potentially related.

I am not sure whether there is a similar problem with MSVC, but this certainly explains missing parts of the response (i.e. `Content-Length` being to small) or the browser endlessly hanging while waiting for the next 5315150 or something bytes that will never arrive.

As opposed to @erappaport's build environment, v6.2 seems to be broken for my MinGW setup as well.

Let me know if I should move this into a separate issue if the cause turns out to be unrelated.

erappaport, 2016-06-08:
I'm now seeing what might be a related issue in v6.2. When allowing users to download files from the website, the system will occasionally not send the entire file. I have not figured out the exact conditions yet, but this is a major problem for my clients.

cpq, 2020-12-13:
Please use the latest code.

closed 2020-12-13 by cpq