cesanta/mongoose #646: freed pointer used in  multithread mode

mccoysc, 2016-04-14:
freed pointer used in  multithread mode.

on  line 4977:
mg_call(nc, (pd->endpoint_handler ? pd->endpoint_handler : nc->handler),
              MG_EV_HTTP_PART_END, &mp);

may be pd have been freed in somewhere when mg_call called.

mccoysc, 2016-04-15:
[image]

rojer, 2016-04-15:
it is a bug that any data race can occur. even in multi=threaded mode one connection is still supposed to only be accessed from only one thread. i'll take a look.

mccoysc, 2016-04-15:
i'v found the key place:

at line 3889:
  c[1]->listener = nc->listener;
  c[1]->proto_handler = nc->proto_handler;
  c[1]->proto_data = nc->proto_data;
  c[1]->user_data = nc->user_data;

if in work thread,i set like this:
c->flags | =MG_F_SEND_AND_CLOSE;

and immediately the main thread found that the MG_F_SEND_AND_CLOSE is set,it will free the conn and the conn's proto_data.
but really,the proto_data is being used by work thread.

so,my resolve is add some code when create work thread:
  c[1]->proto_data_destructor = nc->proto_data_destructor;
  nc->proto_data = NULL;

mccoysc, 2016-04-15:
and also add these code:

  if (c[1]->prev)
  {
	  c[1]->prev= c[1]->next;
  }

  if (c[1]->next)
  {
	  c[1]->next->prev = c[1]->prev;
  }

alashkin, 2016-08-16:
Please, check with latest version

closed 2016-08-16 by alashkin