cesanta/mongoose #656: Connecting using http to ssl port causes mongoose to infinite and CPU usage high

sswork, 2016-05-09:
If mongoose is configure to serve both http and https, then connecting to the https port using http in the browser causes mongoose hang or cpu usage high.

For example, if mongoose is set up to serve http on port 8081 and https on 8082:
http://localhost:8081 ==> works fine
https://localhost:8082 ==> works fine
http://localhost:8082 ==> request goes infinite

* MONGOOSE_VERSION "5.2"

Please help me to resolve this issue.

pjg864, 2016-05-11:
@sswork  I had the same problem.

mmikulicic, 2016-05-11:
This is a quite old version. Can you try to upgrade ?
@sswork, which platform ?

@pjg864 do you also run version 5.2 ?

sswork, 2016-05-23:
@mmikulicic 
Thank you for your reply.

Unfortunately, we could not update the version, as we customized it according to our requirement.
Therefore could you please guide me to resolved this issue?

mmikulicic, 2016-05-23:
Yeah, upgrading to 6.x might be a big effort, but 5.2 is quite old in the 5.x line as well (>2 years).
Please try upgrading to 5.6 or some earlier version.

If it's not possible, then please take a look at the git commit history for commits more recent than
7380c6526594d91b8207449ba39b2befc355d022 (the 5.2 release).

Look for some relevant bug fix, for example 27c665639811574e9d0e737938e1688fd171c1cf might look relevant (fix of SSL accept).

Unfortunately we have no resources to do that archeology work for you; it's a pretty old version and the bug has been fixed long ago. I hope I put you on the right track though.

If you need commercial support feel free to contact our sales https://www.cesanta.com/contact.

closed 2016-05-23 by mmikulicic

sswork, 2016-05-24:
Actually we are using mongoose as a WebDAV.
We want to move/rename WebDAV files by the requests.
does the latest mongoose supports WebDAV server functionality with SSL support?