cesanta/mongoose #672: web_server example is gone

KingDuckZ, 2016-06-17:
Hi, I just noticed that commit 41b405d7bf6e620bf0a8e34e30f135a7e81d613d removes the web_server example, which I was using for my internal website at home. Is there any other project that takes its place? Or is there any way to bring it back? Thanks.

rojer, 2016-06-18:
https://github.com/cesanta/mongoose/tree/master/examples/simplest_web_server

closed 2016-06-18 by rojer

KingDuckZ, 2016-06-18:
I had already looked at that, but it's a trimmed version of [this](https://github.com/cesanta/mongoose/blob/30b80dc5eebd91967438326635ae3a9661956c64/examples/web_server/web_server.c). I can't see how you can change any of the default parameters without recompiling the program.

rojer, 2016-06-18:
look at https://github.com/cesanta/mongoose/tree/master/examples/restful_server then

KingDuckZ, 2016-06-28:
I looked at most examples already, none seems to do the same thing as the removed example (reading a config file, taking command line parameters). That was quite useful on Raspberry Pi for me, so when I get some free time (after I fix my own issues on my current project), I'll think of doing something myself if you don't restore that bit of code in the meantime.