cesanta/mongoose #673: BYTE_ORDER not defined

alecmocatta, 2016-06-20:
```
./mongoose/mongoose.c:1236:5: error: 'BYTE_ORDER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if BYTE_ORDER == BIG_ENDIAN
    ^
./mongoose/mongoose.c:1236:19: error: 'BIG_ENDIAN' is not defined, evaluates to 0 [-Werror,-Wundef]
#if BYTE_ORDER == BIG_ENDIAN
                  ^
./mongoose/mongoose.c:1577:5: error: 'BYTE_ORDER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if BYTE_ORDER == LITTLE_ENDIAN
    ^
./mongoose/mongoose.c:1577:19: error: 'LITTLE_ENDIAN' is not defined, evaluates to 0 [-Werror,-Wundef]
#if BYTE_ORDER == LITTLE_ENDIAN
                  ^
```

I'm not entirely sure what the fix should be. On my system (clang, x86_64-unknown-linux-gnu, glibc), `<endian.h>` defines `__BYTE_ORDER`, `__BIG_ENDIAN`, `__LITTLE_ENDIAN`, but only defines `BYTE_ORDER` etc iff `#ifdef __USE_MISC`, which [seems](http://stackoverflow.com/questions/10231885/what-is-use-misc-macro-used-for/10232115#10232115) to be a BSD thing.

Perhaps mongoose should switch to the `__` prepended versions?

alashkin, 2016-08-24:
Hello, 
Is this problem still actual?
I just tried

GCC:
```
[ ~/go/src/cesanta.com/mongoose @master ] gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.2) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[ ~/go/src/cesanta.com/mongoose @master ] gcc -c mongoose.c
[ ~/go/src/cesanta.com/mongoose @master ] 
```

CLANG:
```
[ ~/go/src/cesanta.com/mongoose @master ] clang --version
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
[ ~/go/src/cesanta.com/mongoose @master ] clang -c mongoose.c
[ ~/go/src/cesanta.com/mongoose @master ] 
```

Could you add more information about your compilation environment/command line used etc?

alecmocatta, 2016-08-24:
@alashkin `clang-3.8 -std=c99 -Werror -Wundef mongoose.c` gives the error I listed.

However, that's irrelevant, the fact is using `BYTE_ORDER`, `LITTLE_ENDIAN`, `BIG_ENDIAN` rather than `__BYTE_ORDER`, `__LITTLE_ENDIAN`, `__BIG_ENDIAN` is incorrect without first defining `__USE_BSD`, see the glibc source here:

https://github.com/lattera/glibc/blob/a2f34833b1042d5d8eeb263b4cf4caaea138c4ad/string/endian.h#L44-L49

alashkin, 2016-08-24:
Fixed in https://github.com/cesanta/mongoose/commit/2deaf084e4dc0214be55b29f964c3d69bbdf6b21

closed 2016-08-24 by alashkin