cesanta/mongoose #674: MQTT QoS > 0

LokiFaun, 2016-06-21:
Hi,

In our company (Swarco Futurit) we are currently trying mongoose mqtt broker and are having problems when publishing messages with a quality of service 1 or 2.

Further investigation has shown that PUBACK (QoS 1,2), PUBCOMP (QoS 2) are never sent. Also message_id is never used.

Furthermore retained messages seem to be unsupported which we would also need.

Is this a bug, not yet implemented or intentional?

mmikulicic, 2016-06-21:
Hi. Do you need broker support or only MQTT client support ?
The current MQTT implementation is very minimal. That can change depending on the interest.

LokiFaun, 2016-06-21:
Hi.
We actually need both.
For the client we currently use paho-mqtt but would like to switch to mongoose to minimize thirdparty dependencies.
For the server we would need mongoose as we haven't gotten any other implementation we tried to even compile for our system (Windows CE 6.3).
Messages with QoS 0 are working fine.

Problem is that we sill also need QoS 1,2 and retained messages depending on our customers.

P.S. I did a few changes for WinCE compatibility which I could provide as a Pull-Request after we verified they are working for us.

mmikulicic, 2016-06-21:
The best way you can speed up progress on this is to talk to our sales team at https://www.cesanta.com/contact and get them interested :-)

w.r.t technical issues, higher QoS require some form of queuing. Do you need persistent queuing ? Do you have append feature on your embedded filesystem ? How much RAM does your target have ?

LokiFaun, 2016-06-21:
My team leader will contact your sales team.

As for your technical questions:
We don't need persistent queuing.
For persisting data we normaly use flash memoy, so I guess we are using FAT filesystem.
Our target system has ~100MB RAM which is currently about 60% in use. Each process can have 32MB RAM at most. I think we are pretty safe here as we don't have have that much topics (currently <20) to be cached or messages to be handled in a short time (<1000 msg/sec)

cpq, 2016-08-25:
@LokiFaun just to clarify, can we safely close this issue?

LokiFaun, 2016-08-25:
Yes. The issue can be closed. We found a way to implement it by calling our own event-handler before calling the default mqtt handler of the library.

closed 2016-08-26 by cpq