cesanta/mongoose #680: udp multicast, broadcast

azadkuh, 2016-07-17:
is it possible to send datagrams via broadcast (and multicast) addresses?
```cpp
mg_connect(mgr, "udp://198.192.10.255:9090", handler);
```
the connection will be connected and then closed immediately under `ubuntu` and `OS X` (i guess i need to set `SO_BROADCAST` somewhere)

the same code just works fine under `windows`.

pingdynasty, 2016-08-23:
What happens if you run as root?

azadkuh, 2016-08-24:
@pingdynasty well, having `su` privilege has no effect on this.
(for example Qt5/QUdpSocket works just fine under normal users)

pingdynasty, 2016-08-24:
Apparently on Linux multicast can be disabled on interface basis.
E.g. to disable on eth0: `ifconfig eth0 -multicast`
to enable: `ifconfig eth0 multicast`

azadkuh, 2016-08-24:
@pingdynasty thank you for your suggestion, be multicast is already enabled on my system(s).

> to be clear, our Qt5 based apps just work, i'm like to replace `QtNetwork` with mongoose due to licensing issue around statically linking Qt5.

alashkin, 2016-09-12:
Fixed in https://github.com/cesanta/mongoose/commit/d9b7ec18e788d1c9cc4aa566a463a21fa58b46e1

Usage example:
```
struct mg_connect_opts opts;
opts.flags = MG_F_ENABLE_BROADCAST;
....
mg_connect_opt(....., opts);
```

closed 2016-09-12 by alashkin

pingdynasty, 2016-09-12:
What about multicast addresses?

alashkin, 2016-09-12:
well, `bind` to 0.0.0.0 should work

azadkuh, 2016-09-12:
@alashkin i think `SO_BROADCAST` is required for broadcasting thank you for the commit.

to write and read from a multicast group (channel) some other socket options may be required as `IP_ADD_MEMBERSHIP / IP_DROP_MEMBERSHIP / IP_MULTICAST_IF / ...`

alashkin, 2016-09-12:
unfortunately `IP_ADD_MEMBERSHIP`  is not just an option like BROADCAST
Ok, we can add parameters for IP_ADD_MEMBERSHIP, but IP_DROP_MEMBERSHIP/IP_MULTICAST_IF etc requires (IMO) additional API.

Anyway, feel free to reopen this issue or write new, more specific, about all multicast stuff

azadkuh, 2016-09-12:
[mulitcast in linux](http://www.tldp.org/HOWTO/Multicast-HOWTO-6.html)

yes, they are not just some simple flags to be set, there are some other minor issues under OS X as `SO_REUSEPORT` is also required for multicasting and I have no idea if they're also available on other (non-BSD-like) stacks.

may be it would be better to open yet another issue (dedicated to multicasting)

snahmad, 2018-08-17:
Does mongoose UDP supports Multicast address joining and sending packets.

I can make broadcast work using mg_send but not by mg_broadcast on widows platform.

on Esp32 platform mg_broadcast id not building. linker error.