cesanta/mongoose #684: Playing with  mg_mgr_init with class instance doesn't work

Pwera, 2016-08-06:
I copied coap client sources and tried to make it as C++ wrapper.
In main.cpp
```
CoapClient* coapClient = new CoapClient();
        coapClient->con(coapClient);
```
https://gist.github.com/Pwera/e2955856bd5d52e30618885c96050852

In void coap_handler(struct mg_connection *nc, int ev, void *p)  method  'ev' parameter has strange value.

alashkin, 2016-08-15:
Hi, 
I'm not sure I understand what do you mean

`nc = mg_connect(&mgr, address.c_str(), CoapClient::Wrapper_To_Call);`

Last parameter or `mg_connect` function is *event_handler* - see here https://docs.cesanta.com/mongoose/master/#/c-api/net.h/mg_connect.md/, function with 3 parameters, not 4 like Wrapper_to_call

`static void Wrapper_To_Call(void* pt2Object, struct mg_connection *nc, int ev, void *p)`

Thus,

``` 
nc = mg_connect(&mgr, address.c_str(), coap_handler);```

...
static void coap_handler(struct mg_connection *nc, int ev, void *p)
....
```

Looks more correct

alashkin, 2016-08-15:
If I understand correctly, you can try something like this:
```
        nc = mg_connect(&mgr, address.c_str(), CoapClient::Wrapper_To_Call);
        nc->user_data = this;
...
       static void Wrapper_To_Call(/* no pt2Obj param! */ struct mg_connection *nc, int ev, void *p) {
            CoapClient* pThis = (CoapClient*)nc->user_data;
            pThis->coap_handler(nc, ev, p);
       }

```

closed 2016-08-16 by alashkin