cesanta/mongoose #687: is there any max tcp connection limit ?

yzsand, 2016-08-18:
i write a test console application to connect mongoose 6.1

when it touch 36 connections, later connection is refused by server

alashkin, 2016-08-18:
Do you mean 37 active connections or just 37th connection is refused?

yzsand, 2016-08-18:
37th

alashkin, 2016-08-18:
I.e. there aren't other active connections?
Are you sure, previous 36 are closed?

yzsand, 2016-08-18:
seems 36 established connections is the  up limit. I didn't test linux version

yzsand, 2016-08-18:
Not that problem. I mean 36 is the max concurrent number.
Impossible, isn't it? Maybe there's another problem cause this

alashkin, 2016-08-18:
>Impossible, isn't it?

Why impossible? We use `seelect` call, on Windows it do have limit

yzsand, 2016-08-18:
But it seems too small,some where to change?

alashkin, 2016-08-18:
Do you build mongoose from sources? Or binaries?
What you app does?
Establish connection to itself? I.e. within the same mg manager?
Could you share this information with me?
Thanx

yzsand, 2016-08-18:
sorry for not say it clear.
I mean mongoose as server app with thread enable 
A standalone tcp console exe as client 
Two applications.

Client send some string by http  post,mongoose server reply the same string.

Both build from source

yzsand, 2016-08-24:
any update info？

alashkin, 2016-09-19:
Hi, sorry for the long wait.
Just tried - works fine for me on Win10 x64 
But I've used current master,
could you try it? There are a lot of bug fixes in it

alashkin, 2016-09-26:
Feel free to reopen, if current master still have this issue

yzsand, 2016-10-18:
@alashkin
i try current master,the problem still exist
what do you mean " works fine",  any concurrent number explain or test play?
can you share me your test plan?

my test scenario:

server:
`
int main(){
	struct mg_mgr mgr;
	struct mg_connection *nc;

	mg_mgr_init(&mgr, NULL);
	nc = mg_bind(&mgr, s_http_port, ev_handler);
	mg_set_protocol_http_websocket(nc);
	mg_enable_multithreading(nc);

	printf("Starting multi-threaded server on port %s\n", s_http_port);
	for (;;) {
		mg_mgr_poll(&mgr, 3000);
	}
	mg_mgr_free(&mgr);
}
`
client : 
        a executable program do 1000 loop of sending http post one by one

action:
start to run client one by one, just to increase concurrent number
then after about 6 clients running, the 7th client will raise a excetion: connection refused by remote server

yzsand, 2016-10-18:
tks for reopen this issue

yzsand, 2016-10-19:
updated test scenario:

server code:

```
static void ev_handler(struct mg_connection *c, int ev, void *p) {
	if (ev == MG_EV_HTTP_REQUEST) {
		struct http_message *hm = (struct http_message *) p;
		char reply[100];

		/* 5 seconds Simulate long calculation to  increase client concurrent number*/
		sleep(5);

		/* Send the reply */
		snprintf(reply, sizeof(reply), "{ \"uri\": \"%.*s\" }\n",
			(int) hm->uri.len, hm->uri.p);
		mg_printf(c, "HTTP/1.1 200 OK\r\n"
			"Content-Type: text/plain\r\n"
			"Content-Length: %d\r\n"
			"\r\n"
			"%s",
			(int) strlen(reply), reply);
	}else if(ev==MG_EV_SSI_CALL){

	}
}

int main(void) {
	struct mg_mgr mgr;
	struct mg_connection *nc;

	mg_mgr_init(&mgr, NULL);
	nc = mg_bind(&mgr, s_http_port, ev_handler);
	mg_set_protocol_http_websocket(nc);
	mg_enable_multithreading(nc);

	printf("Starting multi-threaded server on port %s\n", s_http_port);
	for (;;) {
		mg_mgr_poll(&mgr, 3000);
	}
	mg_mgr_free(&mgr);

	return 0;
}
```

client code use libcurl:

```
static unsigned __stdcall clientAction( LPVOID WorkContext )
{
	ostringstream oss;
	oss << "http://192.168.0.118:8000/test/" << GetCurrentThreadId();

	CURL* pCurl=curl_easy_init();

	struct curl_slist *headers=NULL; 
	headers=curl_slist_append( headers, "Cache-Control: no-cache"); 
	headers=curl_slist_append( headers, "Pragma: no-cache"); 
	curl_easy_setopt (pCurl, CURLOPT_HTTPHEADER, headers);

	curl_easy_setopt (pCurl, CURLOPT_URL, oss.str().c_str());
	curl_easy_setopt(pCurl, CURLOPT_POSTFIELDS, "name=clientAction&project=curl");
	curl_easy_setopt(pCurl, CURLOPT_WRITEFUNCTION, noop_cb);

	long retcode=0;
	CURLcode pCurlRes = curl_easy_perform(pCurl);
	curl_easy_getinfo (pCurl, CURLINFO_HTTP_CODE, &retcode);
	curl_slist_free_all(headers);

[... 257 lines cut ...]
> 223	  TCP    192.168.0.118:8000     192.168.0.118:52325    ESTABLISHED     23400
> 224	  TCP    192.168.0.118:8000     192.168.0.118:52326    ESTABLISHED     23400
> 225	  TCP    192.168.0.118:8000     192.168.0.118:52327    ESTABLISHED     23400
> 226	  TCP    192.168.0.118:8000     192.168.0.118:52328    ESTABLISHED     23400
> 227	  TCP    192.168.0.118:8000     192.168.0.118:52329    ESTABLISHED     23400
> 228	  TCP    192.168.0.118:8000     192.168.0.118:52330    ESTABLISHED     23400
> 229	  TCP    192.168.0.118:8000     192.168.0.118:52331    ESTABLISHED     23400
> 230	  TCP    192.168.0.118:8000     192.168.0.118:52332    ESTABLISHED     23400
> 231	  TCP    192.168.0.118:8000     192.168.0.118:52333    ESTABLISHED     23400
> 232	  TCP    192.168.0.118:8000     192.168.0.118:52334    ESTABLISHED     23400
> 233	  TCP    192.168.0.118:8000     192.168.0.118:52335    ESTABLISHED     23400
> 234	  TCP    192.168.0.118:8000     192.168.0.118:52336    ESTABLISHED     23400
> 235	  TCP    192.168.0.118:8000     192.168.0.118:52337    ESTABLISHED     23400
> 236	  TCP    192.168.0.118:8000     192.168.0.118:52338    ESTABLISHED     23400
> 237	  TCP    192.168.0.118:8000     192.168.0.118:52339    ESTABLISHED     23400
> 238	  TCP    192.168.0.118:8000     192.168.0.118:52340    ESTABLISHED     23400
> 239	  TCP    192.168.0.118:8000     192.168.0.118:52341    ESTABLISHED     23400
> 240	  TCP    192.168.0.118:8000     192.168.0.118:52342    ESTABLISHED     23400
> 241	  TCP    192.168.0.118:8000     192.168.0.118:52343    ESTABLISHED     23400
> 242	  TCP    192.168.0.118:8000     192.168.0.118:52344    ESTABLISHED     23400

the data above means about 241 concurrent

@alashkin 
tks for reading this long post, waiting for your suggestion  :)

alashkin, 2016-10-22:
Since it is Windows, 
Don't you just achieve sockets/select/FD_SET limit?
It is 64 on windows by default. Could you try to increase it?
https://msdn.microsoft.com/en-us/library/windows/desktop/ms739169(v=vs.85).aspx

I tried on my Ubuntu linux, works for ~500 connection and then stops to accept new connections - which i s ok for _my_ system

yzsand, 2016-10-23:
@alashkin Tks ;)
It seems i forget mongoose is based on select mode.
Any IOCP plan on Windows?

closed 2016-10-23 by yzsand

alashkin, 2016-10-23:
Oh.... we've already used 'epoll' for Linux once but later deleted this code  (read the whole story here: https://github.com/cesanta/mongoose/issues/722)
So, I think we don't have any plans for IOCP....

yzsand, 2016-10-23:
That's OK
Thanks for this great project!