cesanta/mongoose #691: multithreaded_restfull_server utilization

thomasknauth, 2016-09-02:
Dear all,

when generating load with wrk2 against a mt_restfull_server, I see multiple threads are spawned. However, their CPU utilization is ~0%, with a single mongoose thread (the main thread) showing a util slightly above 0% (far away from 100%). Overall throughput is low at 3-4K req/sec. Is this expected behavior? wrk2 opens multiple connections and continuously fires requests at mongoose.

With a single threaded mongoose everything works as expected (close to 100% util, ~160K req/sec).

thomasknauth, 2016-09-04:
The problem is related to the size of my reply. This is probably tied to some resource limit on Linux (e.g., how much data you can have sitting in a pipe or whatever).

This makes me wonder why the worker processes do not send their replies directly? Instead everything is proxied by the main thread. Why? The extra copying seems wasteful. Is this to avoid some race conditions/use-after-free sort of things on the mg_connection?

alashkin, 2016-09-07:
>This makes me wonder why the worker processes do not send their replies directly?

Actually they do. Should do.
Each accepted connection sends/receives in its own thread. 
How many requests per second do you generate?

thomasknauth, 2016-09-07:
Thanks for replying!

As far as I understand, the workers use the socketpair created by spawn_handling_thread() to communicate with the main thread. Hence, the main thread relays the entire communication.

I still haven't gotten to the bottom of this. My ev_handler is

```
static void ev_handler(struct mg_connection *c, int ev, void *p) {
  if (ev == MG_EV_HTTP_REQUEST) {
    struct http_message *hm = (struct http_message *) p;
#define SIZE 4096
    char reply[SIZE];
    memset(reply, 'a', SIZE);
    mg_printf(c,
              "HTTP/1.1 200 OK\r\n"
              "Content-Type: text/plain\r\n"
              "Content-Length: %d\r\n"
              "\r\n",
              SIZE);
    mg_send(c, reply, SIZE);
  }
}
```

With a #define SIZE 512 everything works fine. Lots of throughput and high utilization. With 4096 it's crawling at 25 requests/sec. strace -T tells me that the select() calls in the worker take 0.04s to return. This agrees with the 25req/sec. I'm clueless as to why select() takes so long to return though...

thomasknauth, 2016-09-07:
Ah, well. I actually figured it out. Eventually I stumbled upon the default delay for TCP ACKs, which is 40ms. http://stackoverflow.com/questions/2251516/how-can-i-reduce-socket-latency

alashkin, 2016-09-12:
So, should we think what your issue is gone and MG is innocent?

closed 2016-09-12 by thomasknauth