cesanta/mongoose #693: https bug

jinhill, 2016-09-07:
If I set "bind_opts.ssl_ca_cert = s_ssl_ca_cert" for verify client cert, then thorw "ERR_CONNECTTION_CLOSED". can not access the other links.

alashkin, 2016-09-12:
Hi!
Do you mean binding server or client connection?
And what side gets ERR_CONNECTTION_CLOSED - client or server?

jinhill, 2016-09-13:
@alashkin The browser console output error info:" ERR_CONNECTTION_CLOSED".

alashkin, 2016-09-23:
What is `s_ssl_ca_cert` in your case?
Are you sure it is valid CA bundle?
Could you try the same with `curl -vvvv.....` and put its output here?
Thanx!

closed 2016-09-29 by alashkin

jinhill, 2016-10-01:
@alashkin my ca cert is valid. curl cannot keep ssl session.
1st step: use client cert to visit the root directory, this is successful (need verify client cert).
2nd step: visit another link of the root directory, then get ssl error.
may be the SSL connection reuse, the second link how to reuse the SSL handshake?