cesanta/mongoose #695: Support for HTTP/2 ?

patlecat, 2016-09-15:
Do you have any plans to offer support for the upcoming protocol update on HTTP/2? Would be nice to know.

cpq, 2016-09-23:
Yes we plan to implement HTTP/2, but that is not yet in our immediate schedule.
@patlecat can you elaborate more your interest please - are you working on a specific project that can benefit from HTTP/2 support ?

patlecat, 2016-09-26:
I have planned to use mongoose for my project of an application server, it would be serving the HTML output in cases where the users do not want to use Apache or nginx. It is still in the planning phase yet. For users that want to create a webapp, security and speed of delivery will be important, that's why I asked for HTTP/2.

himanshuPatel1973, 2017-06-27:
Does mongoose support HTTP 1.2 protocol

cpq, 2017-06-27:
HTTP/2 is not yet on the roadmap.
Mongoose is primarily used on embedded devices, where low footprint is more important than performance.

closed 2017-06-27 by cpq

himanshuPatel1973, 2017-06-27:
Our application is to develop Apple push notification server and which works  on HTTP 2.0 and TLS 1.2. So that is  what  our  main requirement   to having both in Mongoose. is there  any alternate available to make it work?

cpq, 2017-06-27:
We'd be happy to implement that on the commercial basis. Please contact us at support@cesanta.com if interested.

himanshuPatel1973, 2017-06-27:
ok thanks!