cesanta/mongoose #696: C++ Wrapper?

patlecat, 2016-09-15:
Will you consider offering a C++ wrapper for mongoose? It would be very helpful then I wouldn't have to write my own and not risk to lose track of changes in your code and avoid misunderstandings and hard to fix bugs.

cpq, 2016-09-18:
@alashkin @rojer @mmikulicic shall we do it?

alashkin, 2016-09-18:
I don't think so.
The main question - for what?
Mongoose has quite simple API, if we need to simplify something - we need to do that in C-API as well.
IIRC, Mongoose do not force to use `free()` & Co - i.e. C++ smart pointers won't be useful.
The only reason to add C++ (C#, Java, etc etc) is increasing number of users. But - will it increase users amount?
I'd say C# wrapper might be more useful.

alashkin, 2016-09-18:
@patlecat 

What is your point? What is wrong with C-API?

azadkuh, 2016-09-20:
@alashkin 
c-api is more portable (especially the ABI)
but some `c++11` features/idoms are simply awesome (generic lambda, tmp,  raii, crtp, ...)

ex:
```cpp
struct connection { void echo() {...} };

int
main(int argc, char** argv) {
   ...
    mongoose::manager m;
    m.bind(port, [](mg_connection* pconn, int ev, void*) {
        switch (ev) {
        case MG_EV_ACCEPT:
            new connection(pconn);
            break;

        case MG_EV_RECV:
            mongoose::with<connection>(pconn, [](auto* c) { c->echo(); });
            break;

        case MG_EV_CLOSE:
            mongoose::with<connection>(pconn, [](auto* c) { delete c; });
            break;
       
         default:
             break;
        }
    });

    mongoose_catch_signals({SIGINT, SIGHUP, SIGTERM});
    return m.exec();
}
```
 
`c++` api can be an optional, thin header only (single hpp) wrapper.

kmansoo, 2016-09-21:
I have made C++ wrapper for mongoose in Luna SW Platform.
I think that if you use Luna, you can make applications easier like RESTful API or Websocket service. 

Luna SW Platform: https://github.com/kmansoo/Luna
Sample Application: https://github.com/kmansoo/SimpleLoginRESTfulAPI

NewbProgramming, 2016-09-21:
Why don't you make it yourself? 
Made an Lua wrapper myself, very easy.

patlecat, 2016-09-22:
@kmansoo Thanks I'll have a look at it soon. Hope you always keep it up to date?
@NewbProgramming I don't want to spend time to maintain the wrapper but to spend time to my app
@alashkin It's just a huge time saver when the manufacturer offers an always up to date date with modern C++ (>= 14) and purely a convenience service. Of course I can work with C (with great disgust though) so I didn't say it's a bug not to offer it from your side, but just asking if you would consider it, that's all. Thanks a lot for the mongoose server btw!

patlecat, 2020-03-31:
Is still on the table? It would be good to have it and it's a better time than ever with C++17 features available giving mongoose the extra performance boost and comfortable and safe interface. And watch out for coroutines in C++20 they can revolutionize the performance for sure!

jeremypoulter, 2020-03-31:
I have been working on a C++ library wrapper, [ArduinoMongoose](https://github.com/jeremypoulter/ArduinoMongoose/), and even though it has Arduino in the name it should work just as well for standard C++

cpq, 2020-12-13:
The C++ wrapper could be implemented in many ways. The example above is a nice showcase, but something closer to a real use case would be helpful.

@patlecat could you provide such a close-to-real-life  C++ app code please - for example, take a https://github.com/cesanta/mongoose/tree/master/examples/http-restful-server example and rewrite it in C++?

cpq, 2020-12-25:
Actually, closing this.

The C++ promise-based API would probably be interesting, but that is a big project on its own.
A simple C++ wrapper around C API does not probably have much sense, as it would be a mostly repetitive copy/paste.
If there is a good API suggestion in mind, feel free to reopen this issue.

closed 2020-12-25 by cpq

shmuelie, 2024-01-16:
Things a C++ API would bring over the C API:
- Namespaces, allowing names to not have to all start with `mg_`
- RAII for things like `mg_mgr`
- More type safety, like having an enum class for event types
- Using `std::function` over function pointers
- Having methods on things, instead of having to pass it in as the first parameter all the time.

Like @jeremypoulter I've found myself building a C++ wrapper around mongoose to provide these things and having an official one would be much prefered.

cpq, 2024-01-16:
@shmuelie thank you! could you share your C++ API please? Or provide a couple of samples for client and server in your API?

shmuelie, 2024-01-17:
@cpq you can see me C++ wrapper and usage in https://github.com/shmuelie/mfi-custom-code/tree/main/mfi-server

cpq, 2024-01-17:
@shmuelie thank you! I think some .h files are missing, like in the mg/ folder, could you add them please?

shmuelie, 2024-01-17:
Hmm, I've got nothing local, what do you think is missing?

cpq, 2024-01-17:
@shmuelie ah, headers are in top-level includes/ , I just overlooked them. Thanks for sharing!

shmuelie, 2024-01-18:
@cpq I've pushed some minor changes to the C++ wrapper

shmuelie, 2024-04-08:
@cpq wanted to share that I've worked more on the wrapper and pulled it out of the rest server into https://github.com/shmuelie/mfi-custom-code/tree/main/mgpp