cesanta/mongoose #697: HTTP client not working

OgreTransporter, 2016-09-18:
I've tried the HTTP client example on my local Apache server:
```
http_client --show-headers http://localhost/test.php
```
The PHP script consists of
```PHP
<?php echo "Test"; ?>
```
The result is
```
connect() failed: Unknown error
```

I'm working with commit https://github.com/cesanta/mongoose/commit/806f07db67b624ceb6638a3bf0713286f75e996d. Is it better to use curl for client requests?

cpq, 2016-09-18:
Can you confirm that the browser can successfully browse that URL?

OgreTransporter, 2016-09-19:
I wouldn't create an issue if the browser is not working. IE, Edge, Firefox and Chrome are working. Btw, libcurl is working, too.

alashkin, 2016-09-19:
Could you try 
`http_client --show-headers http://127.0.0.1/test.php`
(127.0.0.1 instead of localhost)?

OgreTransporter, 2016-09-19:
```
http_client --show-headers http://127.0.0.1/test.php
```
Works!

alashkin, 2016-09-23:
Cannot reproduce on my machine :(

A couple of questions:

1. What OS do you use?
2. What `ping localhost`, it is resolves localhost to 127.0.0.1?

OgreTransporter, 2016-09-24:
Windows 7, Microsoft Visual Studio 2013
```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\Transporter>ping localhost

Ping wird ausgeführt für Stephan-PC [::1] mit 32 Bytes Daten:
Antwort von ::1: Zeit<1ms
Antwort von ::1: Zeit<1ms
Antwort von ::1: Zeit<1ms
Antwort von ::1: Zeit<1ms

Ping-Statistik für ::1:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 0ms, Maximum = 0ms, Mittelwert = 0ms

C:\Users\Transporter>nslookup localhost
Server:  fritz.box
Address:  fd00::c225:6ff:fe0f:f345

*** localhost wurde von fritz.box nicht gefunden: Non-existent domain.
```

alashkin, 2016-09-26:
ok, confirmed on Windows

Cause: on linux MG resolves `localhost` via /etc/hosts file, in Windows it doesn't work and Mongoose tries to resolve `hocalhost` via DNS server (Google by default), which returns nothing for localhost

OgreTransporter, 2016-09-26:
> on linux MG resolves localhost via /etc/hosts file, in Windows it doesn't work
This is not complete correct. Windows is also using C:\Windows\System32\drivers\etc\hosts which includes 127.0.0.1 and ::1 for localhost. But it is also hard coded in Windows somewhere to resolve localhost to this addresses. You can remove localhost from host file and you still receive ::1 or 127.0.0.1 for localhost.

alashkin, 2016-09-26:
I didn't mean it is correct at all :)
Btw, on my Win10 localhost commented out by default

OgreTransporter, 2016-09-27:
> Btw, on my Win10 localhost commented out by Default

Yes, but Windows still resolves ::1 or 127.0.0.1 for localhost. I think it's hard coded in WinAPI.

alashkin, 2016-09-27:
You can override `MG_DEFAULT_NAMESERVER`

This code in mongoose:
```
#ifndef MG_DEFAULT_NAMESERVER
#define MG_DEFAULT_NAMESERVER "8.8.8.8"
#endif
```

closed 2016-09-30 by alashkin