cesanta/mongoose #698: multithreaded_restful_server

Cyrus414, 2016-09-19:
I tested the multithreaded_restful_server but it seems can't work. In your code you use sleep(3) to simulate long calculation, I use 3 client to request the server at the same time and the result is, first request wait for 3 seconds, second wait for 6 seconds, third wait for 9 seconds. the server just performs like a single thread server.

alashkin, 2016-09-19:
Hi,
I just tried the following and seems everything is fine
1. In 1st terminal run 
```
[ ~/go/src/cesanta.com/mongoose/examples/multithreaded_restful_server @mg698 ] ./multithreaded_restful_server 
Starting multi-threaded server on port 8000
```

2. In 2nd, 3rd, 4th I run restful client, got:
```
[ ~/go/src/cesanta.com/mongoose/examples/restful_client @mg698 ] date && time ./restful_client 
Mon Sep 19 09:33:16 EEST 2016
Starting RESTful client against http://127.0.0.1:8000
Got reply:
{ "uri": "/" }

real	0m3.002s
user	0m0.000s
sys	0m0.000s
```

```
[ ~/go/src/cesanta.com/mongoose/examples/restful_client @mg698 ] date && time ./restful_client 
Mon Sep 19 09:33:17 EEST 2016
Starting RESTful client against http://127.0.0.1:8000
Got reply:
{ "uri": "/" }

real	0m3.002s
user	0m0.000s
sys	0m0.000s
```

```
[ ~/go/src/cesanta.com/mongoose/examples/restful_client @mg698 ] date && time ./restful_client 
Mon Sep 19 09:33:18 EEST 2016
Starting RESTful client against http://127.0.0.1:8000
Got reply:
{ "uri": "/" }

real	0m3.002s
user	0m0.004s
sys	0m0.000s
```

Could you share more details? Something specific, maybe? Did you try to use mongoose/exampes/restful client for testing?

Cyrus414, 2016-09-19:
Sorry, I use different tabs in one web browser to test the server so it seems not work. When I use different web browsers or the restful client, it works fine. 
Thanks!

alashkin, 2016-09-19:
Oook.... I cannot repro this with browser as well.
I'm closing issue, feel free to reopen it if problem will return

closed 2016-09-19 by alashkin