cesanta/mongoose #705: mg_file_upload_handler

OgreTransporter, 2016-09-27:
I like that function, but is it possible to add an other callback function for own output messages? I like to print some information about the uploaded file to the user. The Default code
```C++
        mg_printf(nc,
                  "HTTP/1.1 200 OK\r\n"
                  "Content-Type: text/plain\r\n"
                  "Connection: close\r\n\r\n"
                  "Ok, %s - %d bytes.\r\n",
                  mp->file_name, (int) fus->num_recd);
```
destroys the html output.

alashkin, 2016-09-27:
I don't think so.
`mg_file_upload_handler` is a helper.
If it is not suitable, you can use MG_EV_HTTP_PART_BEGIN & Co directly, w/out this function.

closed 2016-09-27 by alashkin

OgreTransporter, 2016-09-27:
You're right! This function is nice to see, but not usable because I can't controll the output and there is a bug inside.

OgreTransporter, 2016-09-27:
https://github.com/cesanta/mongoose/blob/master/mongoose.c#L5143
```C++
fus->fp = fopen(fus->lfn, "w");
```
man fopen(3):

> The mode string can also include the letter 'b' either as a last character or as a character between the characters in any of the two-character strings described above. This is strictly for compatibility with C89 and has no effect; the 'b' is ignored on all POSIX conforming systems, including Linux.  (Other systems may treat text files and binary files differently, and adding the 'b' may be a good idea if you do I/O to a binary file and expect that your program may be ported to non-UNIX environments.)

fopen should contain a 'b' or it won't with binary files on windows:
```C++
fus->fp = fopen(fus->lfn, "wb");
```
In case of a text file windows might add some additional invisible bytes to the file. Try to upload a zip file or any other binary file and you'll see that they are not useable.