cesanta/mongoose #706: websocket, Without connection when send data but not fails

criptanet, 2016-09-28:
Hello!
Congratulations on the project.

Bug or ignorance ....

By using the WebSocket API, I'm in a 'while' ... sending data to the server, if the server goes down, I get MG_EV_CLOSE event, perfect.
If I disconnect the ethernet cable customers, is continues to send "sending data websocket" and not aware of that is offline! ?? (I get MG_EV_SEND event...)

How I can see the link status websocket?
It is logical / Normal Behavior?

Thx!

alashkin, 2016-09-29:
Hi,
How many bytes sent reports MG_EV_SEND?

(for MG_EV_SEND last parameter is `int *` with number of sent bytes)

alashkin, 2016-09-29:
And what OS do you use?

criptanet, 2016-09-29:
sent reports  -- MG_EV_SEND = 1330320 bytes?
I send 8010 bytes 

My SO embedded Linux "Linux version 2.6.14"

alashkin, 2016-09-29:
Hm?
I meant
```
static void handler(struct mg_connection *nc, int ev, void *p) {
....
int sent = *(int*)p;
```

And you have 1.3Mb in sent?

criptanet, 2016-09-29:
mmmm but the last parameter is not?

static void ev_handler(struct mg_connection *nc, int ev, void *ev_data) {
  struct websocket_message *wm = (struct websocket_message *) ev_data;
I print wm->size

I see this in a websocket example
but new value is 8024 (It is ok)

alashkin, 2016-09-29:
last parameter contains different information, it depends on event.
In case of MG_EV_SEND it is pointer to int with number of bytes.

Anyway, in this case that means, that OS reported, what send was OK.

It is always reports success? I mean, after disconnection -  how iteration of mg_send (and mg_poll) did you tried?
Might be, your OS requires sometime to close the connection?

criptanet, 2016-09-29:
when the ethernet cable disconnected, MG_EV_SEND tells me that sent the bytes that should send ...

Yes, it could be something from my SO ... but it's a fairly standard linux

inelegant my solution ...
each time i send data, send WEBSOCKET_OP_PING and I hope to receive MG_EV_RECV event. If I get 2 bytes of PONG, I take it for good sending data ...

alashkin, 2016-09-29:
just tried on my Linux desktop - I have the same thing

alashkin, 2016-09-29:
Also, I googled a bit - it is quite normal for linux (and not linux only)
send() will work in this situation until internal buffer is not full

criptanet, 2016-09-29:
Ok...
thx

closed 2016-09-29 by criptanet