cesanta/mongoose #708: Problem with query string in head code

grodansparadis, 2016-09-29:
In the current head code URI's like

http://127.0.0.1:8080/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=1

give an empty query_string in the hm structure while

http://127.0.0.1:8080/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=1&

or

http://127.0.0.1:8080/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain

don't.   

Is this by design or a bug?

alashkin, 2016-09-29:
Just tried current master (simplest_web_server example)
```
 curl "http://127.0.0.1:8000/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=1"
```

Got in hm:
`vscpuser=admin&vscpsecret=secret&format=plain&op=1`

grodansparadis, 2016-09-29:
Strange. I definitively get this result.  One difference in my setup is that mongoose is compiled together with v7 so mongoose is compiled with -DEXCLUDE_COMMON

alashkin, 2016-09-29:
What is your platform/OS?

grodansparadis, 2016-09-29:
Linux, 64-bit

alashkin, 2016-09-29:
Well, I have the same.

Could you try simplest_web_server and check if it works/doesn't work for you?

grodansparadis, 2016-09-29:
Works OK. I will try to do something wimple with v7 + mongoose also and try.

grodansparadis, 2016-09-29:
Hmmm.. seems to work as well.  Strange.

grodansparadis, 2016-09-29:
For

http://127.0.0.1:8080/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=1

I get

GET /vscp/rest HTTP/1.1\\r\\nHost: 127.0.0.1:8080\\r\\nUser-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:45.0) Gecko/20100101 Firefox/45.0\\r\\nAccept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8\\r\\nAccept-Language: en-US,en;q=0.5\\r\\nAccept-Encoding: gzip, deflate\\r\\nCookie: user=admin;

in 

req_len = mg_parse_http(io->buf, io->len, hm, is_req);

for 

http://127.0.0.1:8080/vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=10

I get

io->buf=0x7fffd8001dc0 \"GET /vscp/rest?vscpuser=admin&vscpsecret=secret&format=plain&op=10 HTTP/1.1\\r\\nHost: 127.0.0.1:8080\\r\\nUser-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:45.0) Gecko/20100101 Firefox/45.0\\r\\nAccept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8\\r\\nAccept-Language: en-US,en;q=0.5\\r\\nAccept-Encoding: gzip, deflate\\r\\nCookie: user=admin; session=a9e8e184ba316ee2ba8159d396bb5ef8; original_url=/vscp/\"...

as it should be.

alashkin, 2016-09-29:
Last results - is it with simplest_web_server?
I've wrote in previous comment, what it is work

grodansparadis, 2016-09-29:
No last result is my setup with "10" at the end instead of "1"  ;-)  Very "strange". (does not every programmer miss use that word. The simple... works with and without v7 

I have tried to removed as much as I can but still get the same result.  You can close this issue as it seems it have something to do with my setup anyway then. It has worked in the past at least. I open the issue if I find something related to mongoose.

grodansparadis, 2016-09-29:
Just tested with chrome instead of Netscape and then it works as it should. Funny....

alashkin, 2016-09-30:
Ok, closing

closed 2016-09-30 by alashkin