cesanta/mongoose #710: HTTPS client with Mongoose+Krypton in ARM results in Bus Error

WilsonLP, 2016-10-06:
Hello!

First have to tell HTTP client implemented with Mongoose runs fine in my ARM9.
Problems occur when turned into HTTPS using Krypton. (However Application with HTTPS runs ok in Linux PC)

What is done for HTTPS:
-URLs changed from http:// to https://
-added Krypton files
-built with flags -DMG_ENABLE_SSL=1 & -DMG_DISABLE_PFS=1

Here is some mg debug log:

[image]

Please help.

Cheerio
WilsonLP

cpq, 2016-10-07:
Hi @WilsonLP , can you share your snippet of code that does SSL connection setup, please?
What certificate / private key are you using?

rojer, 2016-10-07:
also, can you please compile with `-DKRYPTON_DEBUG` and post the result here

WilsonLP, 2016-10-07:
Hi,
Yeah no problem. Here is part of my code. Only changed http:// to https:// in url and then normally call mg_connect_http(...)

```
const char *url_post_setup = "https://%s/scale/%s/setup";

static void ev_handler_post_initialize (struct mg_connection *nc, int ev, void *ev_data)
{
    // parse MG_EV_HTTP_REPLY here just for posting command result
    struct http_message *hm = (struct http_message *) ev_data;
    struct mbuf *io = &nc->recv_mbuf;

    switch (ev)
    {
        case MG_EV_CONNECT:
            do_ev_handle_connect (nc, ev, ev_data);
            break;
        case MG_EV_POLL:
            do_ev_handle_poll (nc, ev, ev_data);
            break;
        case MG_EV_CLOSE:
			...
    }
}

static void iot_post_setup (void)
{
    ...
    mg_connect_http (&mgr, ev_handler_post_initialize, url, headers, buf);
    r_tick (&tConnectTime);
    pollFlag = 1;
    while (pollFlag)
    {
        r_tick (&tAct);
        mg_mgr_poll (&mgr, 200);
		// Need to handle timeout here also in addition to after recv MG_EV_POLL
        if (tAct - tConnectTime > CONN_MAX_TIME)
            pollFlag = 0;
    }

```

WilsonLP, 2016-10-07:
And HTTPS server certificate is made with XCA with HTTPS server certificate template.
Signature algorithm is _sha1WithRSAEncryption_
_1024 bit RSA key_
```
X509v3 Basic Constraints critical:
CA:FALSE
X509v3 Subject Key Identifier:
ED:76:BB:56:6A:4A:01:3C:7B:C2:81:DB:F5:AD:61:58:64:4F:BB:7C
X509v3 Key Usage:
Digital Signature, Non Repudiation, Key Encipherment
Netscape Cert Type:
SSL Server
Netscape Comment:
xca certificate
```

I have no HTTPS client cert.

WilsonLP, 2016-10-07:
Here is now more debug output when it works when run inside linux PC

```
iot_cycletimer
IOT post status JSON data:
{
"json_name1":false,
"json_name2":"Powerfail",
"json_name3":0.000000,
"json_name4":2
}

mg_http_common_url_parse 127.0.0.1:443 /device/L1236/status
mg_connect_opt       0xb6900480 127.0.0.1:443   
mg_do_connect        0xb6900480 tcp://127.0.0.1:443
mg_if_connect_tcp    0xb6900480 sock 23 err 0
mg_add_conn          0x8443ac4 0xb6900480
mg_set_ssl2          0xb6900480 ,,
status poll..mg_mgr_poll          select @ 1475842717 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0xb6900480 fd=23 fd_flags=2 nc_flags=24 rmbl=0 smbl=226
ssl_err = 0
state 0 -> 1
kr_send(23, 0xb6900940, 65) = 65
do_recv 0 0
kr_recv(23, 0xb6900b48, 256): -1 11
ssl_err = 2
mg_ssl_begin         0xb6900480 0 res -1 11
mg_ssl_err           0xb6900480 -1 -> 2
do_recv 0 256
kr_recv(23, 0xb6900b48, 256): -1 11
ssl_err = 2
mg_ssl_begin         0xb6900480 0 res -1 11
mg_ssl_err           0xb6900480 -1 -> 2
mg_mgr_handle_conn   0xb6900480 after fd=23 nc_flags=88 rmbl=0 smbl=226
status poll..mg_mgr_poll          select @ 1475842717 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0xb6900480 fd=23 fd_flags=1 nc_flags=88 rmbl=0 smbl=226
ssl_err = 0
do_recv 0 256
kr_recv(23, 0xb6900b48, 256): 256 11
msg 22 len 49, have 251
server hello
num_ciphers 1 num_compressions 1
 + EXT: reneg info
 + client cipher_suite[0]: 0x003c
 + client compression[0]: 0x00
cipher: 0x003c
state 1 -> 5
iret = 1
msg 22 len 652, have 197
shuffle buffer down: 54 consumed, 202 left
do_recv 202 256
kr_recv(23, 0xb6900c12, 54): 54 11
msg 22 len 652, have 251
do_recv 256 256
kr_recv(23, 0xb6900c48, 256): 256 11
msg 22 len 652, have 507
do_recv 512 512
kr_recv(23, 0xb6900d48, 256): 154 11
msg 22 len 652, have 661
certificate
cert 0xb6900b57 642
state 5 -> 6
No cert verification??
iret = 1
msg 22 len 4, have 4
hello done
state 6 -> 7
iret = 1
 + master secret computed
len 16 block_len 16 pad_len 15
state 7 -> 8
kr_send(23, 0xb6900940, 230) = 230
do_recv 0 0
kr_recv(23, 0xb6900d30, 256): -1 11
ssl_err = 2
mg_ssl_begin         0xb6900480 0 res -1 11
mg_ssl_err           0xb6900480 -1 -> 2
do_recv 0 256
kr_recv(23, 0xb6900d30, 256): -1 11
ssl_err = 2
mg_ssl_begin         0xb6900480 0 res -1 11
mg_ssl_err           0xb6900480 -1 -> 2
mg_mgr_handle_conn   0xb6900480 after fd=23 nc_flags=88 rmbl=0 smbl=226
status poll..mg_mgr_poll          select @ 1475842717 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0xb6900480 fd=23 fd_flags=1 nc_flags=88 rmbl=0 smbl=226
ssl_err = 0
do_recv 0 256
kr_recv(23, 0xb6900d30, 256): 91 11
msg 20 len 1, have 86
change cipher spec
iret = 1
msg 22 len 80, have 80
pad ok 15
MAC ok 16
state 8 -> 9
finished (OK)
iret = 1
ssl_err = 0
mg_ssl_begin         0xb6900480 0 res 1 11
mg_if_connect_cb     0xb6900480 connect, err=0
mg_call              0xb6900480 proto ev=2 ev_data=0x8517d98 flags=48 rmbl=0 smbl=226
mg_call              0xb6900480 user ev=2 ev_data=0x8517d98 flags=48 rmbl=0 smbl=226
IOT CONNECT
mg_call              0xb6900480 after user flags=48 rmbl=0 smbl=226
mg_call              0xb6900480 after proto flags=48 rmbl=0 smbl=226
do_recv 0 0
kr_recv(23, 0xb6901040, 256): -1 11
ssl_err = 2
mg_ssl_err           0xb6900480 -1 -> 2
mg_mgr_handle_conn   0xb6900480 after fd=23 nc_flags=112 rmbl=0 smbl=226
status poll..mg_mgr_poll          select @ 1475842717 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0xb6900480 fd=23 fd_flags=2 nc_flags=112 rmbl=0 smbl=226
len 226 block_len 16 pad_len 13
kr_send(23, 0xb6900940, 293) = 293
ssl_err = 0
mg_write_to_socket   0xb6900480 226 bytes -> 23 (SSL)
mg_call              0xb6900480 proto ev=4 ev_data=0x8517da8 flags=48 rmbl=0 smbl=0
mg_call              0xb6900480 user ev=4 ev_data=0x8517da8 flags=48 rmbl=0 smbl=0
IOT MG case default, ev = 4
mg_call              0xb6900480 after user flags=48 rmbl=0 smbl=0
mg_call              0xb6900480 after proto flags=48 rmbl=0 smbl=0
mg_mgr_handle_conn   0xb6900480 after fd=23 nc_flags=48 rmbl=0 smbl=0
status poll..mg_mgr_poll          select @ 1475842717 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0xb6900480 fd=23 fd_flags=1 nc_flags=48 rmbl=0 smbl=0
do_recv 0 256
kr_recv(23, 0xb6901040, 256): 149 11
msg 23 len 144, have 144
pad ok 13
MAC ok 82
iret = 1
ssl_err = 0
mg_handle_tcp_read   0xb6900480 82 bytes <- 23 (SSL)
mg_recv_common       0xb6900480 82 0
mg_call              0xb6900480 proto ev=3 ev_data=0x8517d84 flags=48 rmbl=82 smbl=0
mg_call              0xb6900480 user ev=3 ev_data=0x8517d84 flags=48 rmbl=82 smbl=0
IOT MG case default, ev = 3
mg_call              0xb6900480 after user flags=48 rmbl=82 smbl=0
mg_call              0xb6900480 user ev=102 ev_data=0x8517a40 flags=48 rmbl=71 smbl=0
IOT MG case default, ev = 102
mg_call              0xb6900480 after user flags=48 rmbl=71 smbl=0
mg_call              0xb6900480 user ev=101 ev_data=0x8517a40 flags=48 rmbl=71 smbl=0
DBG 3:nothing
IOT NO Command
IOT http response: { "command": "nothing" }, msg len 24 body len 24 io len 71
mg_call              0xb6900480 after user flags=2096 rmbl=0 smbl=0
mg_call              0xb6900480 after proto flags=2096 rmbl=0 smbl=0
mg_ssl_err           0xb6900480 82 -> 0
ssl_err = 0
mg_mgr_handle_conn   0xb6900480 after fd=23 nc_flags=2096 rmbl=0 smbl=0
mg_close_conn        0xb6900480 2096
mg_call              0xb6900480 proto ev=5 ev_data=(nil) flags=2096 rmbl=0 smbl=0
mg_call              0xb6900480 user ev=5 ev_data=(nil) flags=2096 rmbl=0 smbl=0
IOT ev_handler_post_status: MG_EV_CLOSE.
mg_call              0xb6900480 after user flags=2096 rmbl=0 smbl=0
mg_call              0xb6900480 after proto flags=2096 rmbl=0 smbl=0
```

WilsonLP, 2016-10-07:
Here debug output when it crashes

```
iot_cycletimer
IOT post status JSON data:

{
"jsonname1":false,
"jsonname2":"A2909-207:Application terminated",
"jsonname3":1000.000000,
"jsonname4":607
}

mg_http_common_url_parse 192.168.1.117:443 /scale/L1234/status
mg_connect_opt       0x4697b8 192.168.1.117:443
mg_do_connect        0x4697b8 tcp://192.168.1.117:443
mg_if_connect_tcp    0x4697b8 sock 37 err 0
mg_add_conn          0x39e5a4 0x4697b8
mg_set_ssl2          0x4697b8 ,,
status poll..mg_mgr_poll          select @ 1475856228 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0x4697b8 fd=37 fd_flags=2 nc_flags=24 rmbl=0 smbl=252
ssl_err = 0
state 0 -> 1
kr_send(37, 0x445348, 65) = 65
do_recv 0 0
kr_recv(37, 0x44ecb8, 256): 256 22
msg 22 len 49, have 251
server hello
num_ciphers 1 num_compressions 1
 + EXT: reneg info
 + client cipher_suite[0]: 0x003c
 + client compression[0]: 0x00
cipher: 0x003c
state 1 -> 5
iret = 1
msg 22 len 652, have 197
shuffle buffer down: 54 consumed, 202 left
do_recv 202 256
kr_recv(37, 0x44ed82, 54): 54 22
msg 22 len 652, have 251
do_recv 256 256
kr_recv(37, 0x445650, 256): 256 22
msg 22 len 652, have 507
do_recv 512 512
kr_recv(37, 0x445750, 256): 154 22
msg 22 len 652, have 661
certificate
cert 0x44555f 642
state 5 -> 6
No cert verification??
iret = 1
msg 22 len 4, have 4
hello done
state 6 -> 7
iret = 1
 + master secret computed
Caught 7 (Bus error)

```

WilsonLP, 2016-10-07:
Does the wireshark capture give any hint what could be wrong?

[image]

rojer, 2016-10-07:
thanks Wilson, clearly something inside Krypton goes wrong. i will need to take a closer look.
could you attach the wireshark pcap file here?

rojer, 2016-10-07:
also, which library is used on the other end?

WilsonLP, 2016-10-10:
Pcap file sent by email.
On the other end ie. my HTTPS test server is just a RESTful server taken from Mongoose examples running with Mongoose + Krypton.

rojer, 2016-10-10:
ok, thank you. so the crash happens when constructing a reply to server's hello.
SIGBUS may be the result of unaligned memory access. i'm still looking, but do you know if your target CPU allows unaligned fetches? i think it's an option on ARM that can either be enabled or disabled.

WilsonLP, 2016-10-10:
Unaligned memory access Is disabled (and can not / should not be enabled). With some cost of performance one could use “#pragma pack (1)” to create a struct without any padding.

rojer, 2016-10-10:
yes, i agree, alignment errors should be fixed. i just want to track it down, but i don't have access to an ARM board that will run krypton and give me bus errors on unaligned access (rpi won't do it, even with proper /proc/cpu/alignment mode).
can you put a dprintf statement (here)[https://github.com/cesanta/krypton/blob/b2a1889/krypton.c#L6628], before tls_send to see if we get through RSA encryption.

WilsonLP, 2016-10-10:
I put

`dprintf(("RSA encrypt OK\n")); `

On line 6628 and then compiled and run and I get "RSA encrypt OK" printout just before message "Caught 7 (Bus error)"

rojer, 2016-10-10:
ok, now [line 6386](https://github.com/cesanta/krypton/blob/b2a1889/krypton.c#L6386) please.
we do have an upper limit because w know message at line 6415 is not printed.

i'm trying to find suitable hardware so i can do it locally, but so far - no luck.

WilsonLP, 2016-10-11:
Hello world,

After adding some (or many to be honest) drpintfs + one counter variable I can see it crashes on line 4133 when it is executing that function 'SHA256_Transform' 78th time. 

Code:
```
void SHA256_Transform(SHA256_CTX *context, const sha2_word32 *data_in) {
    static int nWilson = 1;
    dprintf(("RSA encrypt OK2.952\n"));
  sha2_word32 data_c[SHA256_BLOCK_LENGTH / sizeof(sha2_word32)];
  sha2_word32 a, b, c, d, e, f, g, h, s0, s1;
  sha2_word32 T1, T2, *W256;
  sha2_word32 *data = data_c;
  int j;
  dprintf(("RSA encrypt OK2.9521, nWilson:%d\n", nWilson++));

  MEMCPY_BCOPY(data_c, data_in, sizeof(data_c));
  dprintf(("RSA encrypt OK2.953\n"));
  W256 = (sha2_word32 *) context->buffer;
```

Output:
```
RSA encrypt OK2.951

RSA encrypt OK2.952

RSA encrypt OK2.9521, nWilson:78

Caught 7 (Bus error)

```
???

rojer, 2016-10-11:
so it crashes in MEMCPY_BCOPY? hm. can you modify the 2.9521 printf to:
```
dprintf(("RSA encrypt OK2.9521, nWilson:%d %p %p\n", nWilson++, data_c, data_in))`
```

WilsonLP, 2016-10-13:
```
RSA encrypt OK2.951

RSA encrypt OK2.952

RSA encrypt OK2.9521, nWilson:78 0x3f9e20 0x3fa12b

Caught 7 (Bus error)
```

rojer, 2016-10-13:
ok, so `data_in` is misaligned.
however, this memcpy (MEMCPY_BCOPY resolves to memcpy) is done precisely to put the data into buffer that is properly aligned (data_c is an array of 32-bit integers and thus has to be aligned - and it as, as we can see from the output above).

so, to me it seems like a buggy implementation of memcpy that cannot handle unaligned copies. can you verify that `memcpy` works with unaligned pointers? it can internally optimize transfer by using larger words, but in the end it is supposed to operate on byte pointers, so should be able to handle unaligned source and/or destination.

WilsonLP, 2016-10-14:
Yes I am quite sure memcpy can work with byte pointers also in my platform. I have done it in other place I think.

rojer, 2016-10-14:
well, there isn't much room for anything else to fail there... i mean, we don't reach printf 2.953.
source pointer is odd, signal is sigbus, which is what happens on unaligned access.

can you verify?

something like this:
```c
int foo[3];
char bar[64];
memcpy(&foo, bar + 1, 32);
```

WilsonLP, 2016-10-14:
Verified.
Misaligned access using memcpy works. Even after typecasting it to (int *) (which krypton does).
I will get bus error if I assign it directly `foo[0] = *(int *)pbar;`

But how Krypton crashed, I don't know?

```
#if 1    // TEST
    int foo[8];
    char bar[64];
    char *pbar = bar;
    pbar += 1;
    thr_printf("TEST 1.000, foo %p  bar %p pbar %p\n", foo, bar, pbar);
    // read unaligned using memcpy
    memcpy(foo, pbar, 32);
    thr_printf("TEST 1.001\n");
    // read unaligned using memcpy
    memcpy(foo, (int *)pbar, 32);
    thr_printf("TEST 1.002\n");
    // access unaligned 2
    foo[0] = *(int *)pbar;
    thr_printf("TEST 1.003\n");
#endif    // END TEST
```

Output:
TEST 1.000, foo 0x3fae28  bar 0x3fade8 pbar 0x3fade9
TEST 1.001
TEST 1.002
Caught 7 (Bus error)

rojer, 2016-10-25:
hi Wilson, sorry for the delay. however, i've not been ignoring the issue, to the contrary - i decided to use the result of my NXP K34 port that i've been working on as a platform to try to reproduce this issue.
that is now done, Mongoose and Krypton have been ported to K34 (see [this article](https://www.hackster.io/rojer/mongoose-embedded-networking-library-on-nxp-kinetis-b5b881) if you're curious). as most ARM implementations these days, it has unaligned access enabled by default, but i specifically enabled traps on unaligned access and did all my work with that mode enabled (i also had to add `-mno-unaligned-access` because it seems to be enabled by default in gcc-arm)
i did end up making a bunch of fixes to Krypton (see [last few commits](https://github.com/cesanta/krypton/commits/master)), but none that i think would be related. i could not reproduce the bug in my testing, kryptong was able to complete handshake and send/receive data in client and server mode, with krypton or openssl on the other side.
so, unfortunately, the cause of the problem remains a mystery to me.
one suggestion: can you see if the compiler is configured to avoid unaligned access?
assuming you're using gcc-arm, you can test it by putting this anywhere in your program:

```c
#if __ARM_FEATURE_UNALIGNED
#error Unaligned access is enabled
#endif
```

if it errors out, then the toolchain is not configured to produce code that respects alignment constraints, assuming that processor supports unaligned loads and stores. as i said, i ran into that in my initial tests and had to explicitly specify `-mno-unaligned-access`.

WilsonLP, 2016-11-04:
Hi and thanks rojer!

Finally I had time to continue with this. I took the latest Krypton master and tried it.
Unfortunately **still getting Bus error**.

I am only able to make it work in my target CPU if I select "Fix the problem and report" instead of "Send SIGBUS and report" in my linux configurator.

[image]

WilsonLP, 2016-11-04:
```
mg_ev_mgr_init       0x39d01c using select()
mg_mgr_init          ==================================
mg_mgr_init          init mgr=0x39d01c
iot_cycletimer
IOT post setup JSON data:
{
"foo":"bar"
}

mg_http_common_url_parse 192.168.1.117:443 /nnn/L2000/setup
mg_connect_opt       0x44d988 192.168.1.117:443
mg_do_connect        0x44d988 tcp://192.168.1.117:443
mg_if_connect_tcp    0x44d988 sock 37 err 0
mg_add_conn          0x39d01c 0x44d988
mg_set_ssl2          0x44d988 ,,
setup poll..mg_mgr_poll          select @ 1478281016 num_ev=1 of 1, timeout=200
mg_mgr_handle_conn   0x44d988 fd=37 fd_flags=2 nc_flags=24 rmbl=0 smbl=208
mg_ssl_begin         0x44d988 0 res -1 11
mg_ssl_err           0x44d988 -1 -> 2
Caught 7 (Bus error)

```

rojer, 2016-11-04:
hm. clearly, something about your setup is different to the one i used.
can you get a debugger connected to this target? i'd be very interested to see where the actual fault happens.

rojer, 2016-11-04:
we've been using Krypton on a platform with strict alignment requirements - Xtensa lx106 (ESP8266) - for months now, and as i said, NXP K64 in strict mode works for me too, so i'm really puzzled. best bet would be to get it on a debugger, and ideally set up a call where i could poke around.

WilsonLP, 2016-11-04:
Ok. I believe you. Is it possible there is something wrong in my server certificate?

rojer, 2016-11-04:
unlikely. however, to rule that out - try test certs from https://github.com/cesanta/krypton/tree/master/test
server.{pem,key} are the server cert and key, ca.pem is the ca cert to verify server.pem

WilsonLP, 2016-11-04:
Ok, thanks. I will try those when I am back in the office - at latest on Monday.

rojer, 2016-11-21:
having looked at this with you over vc, i am pretty convinced that this is a bug in memcpy implementation on the target platform.

closed 2016-11-21 by rojer