cesanta/mongoose #713: how to copy mongoose.c and mongoose.h files to my build tree?

Guguant, 2016-10-10:
When I try to build big_upload, error follows:

`big_upload.c: In function ‘handle_upload’:
big_upload.c:42:10: error: ‘MG_EV_HTTP_PART_BEGIN’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_BEGIN: {
          ^
big_upload.c:42:10: note: each undeclared identifier is reported only once for each function it appears in
big_upload.c:60:10: error: ‘MG_EV_HTTP_PART_DATA’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_DATA: {
          ^
big_upload.c:71:10: error: ‘MG_EV_HTTP_PART_END’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_END: {
          ^
sunyiming@sunyiming:~/桌面/big_upload$ gcc -o main big_upload.c 
big_upload.c: In function ‘handle_upload’:
big_upload.c:42:10: error: ‘MG_EV_HTTP_PART_BEGIN’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_BEGIN: {
          ^
big_upload.c:42:10: note: each undeclared identifier is reported only once for each function it appears in
big_upload.c:60:10: error: ‘MG_EV_HTTP_PART_DATA’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_DATA: {
          ^
big_upload.c:71:10: error: ‘MG_EV_HTTP_PART_END’ undeclared (first use in this function)
     case MG_EV_HTTP_PART_END: {
          ^
`

<font color = "#FF0000"> When I create a project, how to link mongoose.h and mongoose.c ? :-)</font>

alashkin, 2016-10-10:
Hi!

>When I try to build big_upload, error follows:

Do you have these errors then run `make` in examples/bin_upload folder? Did you change example or it is unchanged one?

Guguant, 2016-10-10:
Good evening.
in mongoose/example/ folder, all example run prefect. But when I move bin_upload folder to /home/desktop/, then copy mongoose.h mongoose.c to the /home/desktop/bin_upload, next to command： `gcc -o bin_load bin_upload.c mongoose.c -lpthread `.

I try add
`
#define MG_EV_HTTP_PART_BEGIN 122
#define MG_EV_HTTP_PART_DATA 123
#define MG_EV_HTTP_PART_END 124
`
to big_load.c

Amazing,  127.0.0.1:8000 page is ok.  But when I upload file, it still stay in 127.0.0.1:8000 not 127.0.0.1:8000/upload.

rojer, 2016-10-10:
file upload feature is not enabled by default. you need to compile mongoose with `-DMG_ENABLE_HTTP_STREAMING_MULTIPART`, which is what the example [does](https://github.com/cesanta/mongoose/blob/master/examples/big_upload/Makefile#L2).

Guguant, 2016-10-10:
Thank you @rojer . I have tried to your solution. But issue still exists.

### this is my steps.
[image]

**1. add #define ***   to big_upload.c** 
``````java
15 #define MG_EV_HTTP_PART_BEGIN 122
16 #define MG_EV_HTTP_PART_DATA 123
17 #define MG_EV_HTTP_PART_END 124
```
![2016-10-10 23-10-41](https://cloud.githubusercontent.com/assets/12713192/19241522/9ce7f1a8-8f41-11e6-93da-729da263aa2a.png)

**2. Makefile content:**
````Makefile
  1 objects = big_upload.o mongoose.o
  2 
  3 main:$(objects)
  4         gcc -o main $(objects) -DMG_ENABLE_HTTP_STREAMING_MULTIPART
  5 
  6 $(objects): mongoose.h
  7 
  8 .PHONY: clean
  9 clean:
 10         -rm main $(objects)

```

**3. Run the server**
`./main`
`Starting web server on port 8000`

Finally, the page URL  is staying at 127.0.0.1:8000. Very Slowly.

![2016-10-10 23-07-46](https://cloud.githubusercontent.com/assets/12713192/19241651/2ac9d838-8f42-11e6-87b4-425369257687.png)

***KEY. I want to know Why the page can't jump to 127.0.0.1/upload as expected?***
THANK YOU.

alashkin, 2016-10-10:
Why do you still need step 1?
```
1. add #define *** to big_upload.c
```
These definitions already exists in mongoose.h

alashkin, 2016-10-10:
Makefile doesn't look correct

I don't see compilation of mongoose

What about this?

```
sources = big_upload.c mongoose.c
 
main:$(source)
		gcc $(sources) -o main -DMG_ENABLE_HTTP_STREAMING_MULTIPART
```

Guguant, 2016-10-10:
OMG. succeed. @alashkin you are right.

Thanks very much. I  understand.  o(∩∩)o...

(NOW 00:15am in Beijing) good night.