cesanta/mongoose #714: sending data from websocket client is sometimes not read for long time

heroic, 2016-10-10:
At times, it seems mongoose gets stuck on reading from a websocket client. As in, it's not stuck(I can reload the client, and it gets the new client, and also processes a message or two), but it stops responding after one or two messages. Then it suddenly proccesses another 2-3 messages before getting stuck. At times the stuck time is as long as 5 hours.

Why could this be happening? I'll have to arrange for a working demo if you need one, or you could see the sample at https://github.com/heroic/musika

heroic, 2016-10-10:
Ok, so I have narrowed down the problem to this. When I am trying to mg_send_websocket_frame, and if that connection is dead, then, mongoose never times out. It just remains stuck trying to send data to this client.

alashkin, 2016-10-17:
Looks like https://github.com/cesanta/mongoose/issues/706 ?

alashkin, 2016-10-24:
Closed due to inactivity

closed 2016-10-24 by alashkin