cesanta/mongoose #717: ESP32 porting and compilation

nkolban, 2016-10-14:
I have managed to port Mongoose to run on the ESP32.  There were some issues related to that and some source changes were needed.  This issue is being raised to see if and/how you would like those tracked?  Do you want one issue (this one) for all ESP32 changes needed?  The issues appear to be related to ESP32 environment specific but would have ramifications across the board.  For example ... the biggest I have come across is the inability to format or parse int64_t.  This manifests in the source of mongoose as neither:

```
printf("%ld\n", myInt64);
```
nor
```
printf("%lld\n", myInt64);
```
will work.

The solution would appear to be to not to use 64 bit integers or use "%ld" and cast to unsigned longs where needed.  On the ESP32, an unsigned long is 64 bits.

rojer, 2016-10-14:
this is an issue with ESP build of newlib, which is built without support for 64-bit ints or floats.
the same issue exists on ESP8266. it's a questionable decision on their part, and the preferred solution, for us, has been to [enable](https://github.com/cesanta/esp-open-sdk/commit/c18ea05dfb90d4a15c6517a1bfeda3892f7e25dd) support for these formats in newlib.

alashkin, 2016-10-24:
@rojer 

Shall we close this issue?

rojer, 2016-10-24:
@nkolban, apart from printf issues, was there anything else interesting?

nkolban, 2016-10-24:
What I did was clone a copy of the project and sanbox (hack it) till it worked.  My plan was to try and get it working in a couple of evenings and if that was successful, archive it away and re-clone a virgin version and methodically go through change by change what I found I had to do to get it working.  The printf issues took the lions share of the time.   I believe I also had "linker" battles.  Since raising the printf issue ... I stepped away pending the discussions.  I started a thread in ESP32 land with the belief that I was doing something wrong and the missing function we need was present.  After y'all raised the issue with Espressif ... I stepped away some more to see how that would resolve ... and that's where we are.  Functionally, all seems to be working ... I have tested HTTP and MQTT parts.  I haven't tried Web Sockets yet.   The next big hole is looking into "mbed TLS" (aka POLAR) as the TLS layer as we will need to get TLS working to be useful.

So in summary:

* printf of large int64_t
* linking instructions
* Makefile changes
* ESP32 "component" architecture building
* Support of "mbed TLS"
* Testing across the board of all functions

The question before us is whether y'all want to provide support for ESP32?  Whether you want to do that in private and then release or whether you want community participation?  And how are we to resolve the "conflicts" between what Mongoose currently needs vs what ESP32 currently provides?

For example ... in Mongoose today, we seem to use int64_t.  What was the thinking behind this?  Are int64_t types exposed in the public APIs?  Can we re work the code base to use uint32_t as opposed to 64bit type?

rojer, 2016-10-24:
the "thinking" behind using int64_t is that on larger platforms (think desktop) sizes and offsets > 2G are possible. if was natural, there wasn't actually much thinking involved :) other "smaller" platforms (ARM-based) seem to have no problem with working with 64 bit numbers, ESP8266 (and now 32) are actually a small minority in this regard. while it's possible to get rid of 64-bit ints form mongoose codebase, it's undesirable - we tried before, it's a mess. we'd rather ESP just supported them properly, and they seem to agree - we've been promised a fix on the linked bug.

we will definitely support ESP32 in the near future.

alashkin, 2016-11-01:
@rojer: closing?

rojer, 2016-11-01:
@nkolban, we are open to contributions that will improve (add?) ESP32 support.
however, of the list you posted, other than the int64_t issue, they don't seem to apply to mongoose.{c,h}.
do you have other changes to Mongoose itself?

nkolban, 2016-11-01:
Not at this time ... will close at your request.  Thank you for tracking the int64_t issue.

closed 2016-11-01 by nkolban