cesanta/mongoose #721: error: declaration of 'long long int strtoll(const char*, char**, int)' has a different exception specifier

peku33, 2016-10-17:
Hi,

When I try to compile a c++ object with mongoose.h included, I get the following error:
```
In file included from /.../HttpWebSocketThreadedServer.cpp:1:0:
Libraries/mongoose/mongoose.h:337:45: error: declaration of 'long long int strtoll(const char*, char**, int)' has a different exception specifier
 long long strtoll(const char *, char **, int);
                                             ^
In file included from Libraries/mongoose/mongoose.h:322:0,
                 from /.../HttpWebSocketThreadedServer.cpp:1:
/usr/include/stdlib.h:209:22: error: from previous declaration 'long long int strtoll(const char*, char**, int) throw ()'
 extern long long int strtoll (const char *__restrict __nptr,
```
`strtoll` was introduced with noexcept in c++11.
My solution was to replace mongoose.h:336

from
`#if !(defined(__DARWIN_C_LEVEL) && __DARWIN_C_LEVEL >= 200809L)`
to
`#if !(defined(__cplusplus) && __cplusplus >= 201103L) && !(defined(__DARWIN_C_LEVEL) && __DARWIN_C_LEVEL >= 200809L)`

With this both C and C++11 compiler can compile objects with mongoose.h

alashkin, 2016-10-18:
Hi,
What OS and compiler do you use?

peku33, 2016-10-18:
Gentoo Linux, GCC 4.9.3

closed 2016-10-22 by rojer