cesanta/mongoose #722: I have a question: is Mongoose works fast?

dedok, 2016-10-18:
Hello there!

I'm Vasily, from the tarantool team (github.com/tarantool/tarantool)
i've looked in the mongoose sources and it looks great. guys thanks for that!
But I wondering have somebody have result of comparison mongoose vs NginX?
Those results can be very useful for my plans.

My plan is use mongose into tarantool.

dedok, 2016-10-18:
Sorry for the question, I've saw that the mongoose use select for polling.
That wouldn't work in my case.

cpq, 2016-10-19:
https://habrahabr.ru/post/207460/ - это пост 2013 года с бенчмарком.

dedok, 2016-10-19:
Спасибо за ссылку! А есть планы сделать возможность использовать epoll и т.п.?

cpq, 2016-10-19:
У нас epoll уже был на linux, но мы его убрали.
Причина: основной use case - это embedded, где мангуст крутится на слабых загрузках. @rojer , ты не помнишь что там не так еще с epoll было?

@dedok , в вашем случае - много висящих соединений?

dedok, 2016-10-19:
@cpq Ага, очень много. Сейчас мы используем свой NginX-модуль(https://github.com/tarantool/nginx_upstream_module) для этого, но есть проблема с WebSocket, в NginX сделать WebSocket с upstream модулем очень сложно.
Из-за этого у меня есть мысль использовать мангуст, как buildin web-server - я глянул код, добавить epoll не очень-то и сложно, вот думаю, а не сделать ли мне это.

dedok, 2016-10-20:
@cpq Привет!
А если я добавлю поддержку epoll, то могу я рассчитывать что это будет принято в основную ветку?

dedok, 2016-10-20:
I've reopened it for awhile, since I wanna get an answer :)

rojer, 2016-10-20:
hi Vasiliy.

i was the one who [added](https://github.com/cesanta/fossa/commit/3a54681ff30542fe1b9075b77f1b66b5f775be64) epoll() support (to what back then was still called Fossa) and i was the one who [removed](https://github.com/cesanta/mongoose/commit/19dd02acfd443f12ce278e7d269db872393a7f4f) it.
the latter commit gives the reasons why, the most important of them is actually (3) - with Mongoose, we do not value performance as much as portability, simplicity and ease of maintenance.
we do not compete and will never outperform Mongoose, Apache or Varnish and we are fine with that.
maintaining epoll() proved to be more of a problem than it's worth to us. it was also not as effective - reason (2) states it briefly, but because of the semantics of MG_EV_POLL we _have_ to walk the connection list anyway, so even with epoll we will still scale linearly, the difference is in the constant factor (no need to walk the list beforehand. so, epoll provides limited benefit (in our circumstances) to what is not really important to us. that's why the support was removed and i'd be reluctant to bring it back. not while `MG_EV_POLL` exists. i'm not a fan, but it's been there since forever, users have come to depend on it and we cannot just drop it, not without providing some sort of replacement. one might argue that timer events can be used for that - if you need regular polling, set a timer (maybe persistent).
anyway, it's not a priority for us.

alashkin, 2016-10-24:
I believe @rojer ave an answer, thus, I'm closing issue

closed 2016-10-24 by alashkin

dedok, 2016-10-24:
Yep, Sorry I forgot to close the issue. Thanks you guys!