cesanta/mongoose #723: Can mongoose be made to work with mbed TLS?

nkolban, 2016-10-20:
I understand that there is a TLS/SSL stack called "mbed TLS".  See the following:

https://tls.mbed.org/

I am assuming that mongoose today does not work with this stack?  If not, can it be made to?

The back story is that the new ESP32 ESP-IDF framework provides "mbed TLS" as the TLS/SSL stack and I'd like to make secure https requests and MQTT requests through mongoose and believe that support of "mbed TLS" would be required.

WilsonLP, 2016-10-20:
I studied SSL libraries a bit recently. mbed TLS does not have OpenSSL API so _at the moment_ it cannot be used.

Cheerio, Wilson.

mmikulicic, 2016-10-20:
mbed TLS used to be called polarSSL. 
Back in 2015, we created a wrapper for polarSSL that gives it the OpenSSL API and allows it thus to be used by mongoose: https://github.com/cesanta/polar

That library wraps polarSSL/mbedtls and it was tested with mbedtls 1.3.10.

Did the mbed TLS API change much since then?

nkolban, 2016-10-25:
I have tested out the mbed TLS libraries on an ESP32 and have gotten my mind around them somewhat.  My next task will be to test the polar library you referenced in your last post and see how far I get with that with respect to mongoose on ESP32.

mmikulicic, 2016-10-25:
we are succesfully using our Krypton (https://github.com/cesanta/krypton) TLS library on esp8266 and other embedded platforms. It should work on esp32 as well. Have you considered it?

nkolban, 2016-10-25:
Thanks ... I didn't know about krypton.  Currently I am studying ESP32 for the purposes of  book writing.  Espressif are distributing mbedTLS with their ESP-IDF development framework.  As such, embedTLS seems to be "working out of the box" on an ESP32 ... but I'll look at Krypton and see if it too will work.

alashkin, 2016-10-28:
@nkolban 

Can I close this issue?

closed 2016-10-28 by nkolban