cesanta/mongoose #725: load_balancer.c Segmentation fault under seige

shsssc, 2016-10-25:
I did a stress test. Put the load_balancer example under siege. When I stop the Siege process, the load_balancer process stopped
>\# ./load_balancer -p 8080 -b / 138.68.55.1:80
>Adding backend for / : 138.68.55.1:80 [redirect=0,prefix_replacement=/]
>Starting LB on port 8080
>Segmentation fault

cpq, 2016-10-26:
Could you elaborate on your environment please?

shsssc, 2016-10-26:
I am running on kali2 
>\# uname -a
Linux **** 4.7.0-kali1-amd64 \#1 SMP Debian 4.7.6-1kali1 (2016-10-17) x86_64 GNU/Linux

I also tested on ubuntu 16 and it worked well
It doesn't look like a common problem
It's weird that the program crash only when I stop siege, instead of under it

cpq, 2016-10-27:
@shsssc we have a new API for making reverse proxying, with relevant example.
@alex let's kill this example and close this bug

alashkin, 2016-11-01:
Example deleted

closed 2016-11-01 by alashkin