cesanta/mongoose #727: Mislead about examples without dealing MG_EV_CLOSE

osDanielLee, 2016-10-27:
After connect with server is interrupted due to server down, client will close the connect and call ev as MG_EV_CLOSE, but doesn't deal it in example make it will be easy to ignore. It'll raise a big problem which client point will hang in mg_mgr_poll because mg_mgr_poll always run in a while loop like:
`
  while (s_exit_flag == 0) {
    mg_mgr_poll(&mgr, 1000);
  }
`
Need to deal with MG_EV_CLOSE to change s_exit_flag to break the loop.
Advise to add some code in examples or reference document to give a detailed explanation.

cpq, 2016-10-27:
Makes sense.
@alex could you add `MG_EV_CLOSE` handling to our client examples, and amend the doc, please?

osDanielLee, 2016-10-31:
I've committed an example as #728(https://github.com/cesanta/mongoose/pull/728), maybe you guys could take a look, I'll be very pleasure if I could help you and @alex to complete all client examples and doc.

alashkin, 2016-11-01:
Fixed in https://github.com/cesanta/mongoose/commit/e62dc8f3b2419d4ba25ec9706600b056da0d2fdc

closed 2016-11-01 by alashkin