cesanta/mongoose #731: Extra headers do not work

sibputnik, 2016-10-29:
Hi

I set in config extra_headers parameter, but in servers response it is absent. I have reboot Mangoose, but result is the same. I set other headers, but result the same. 
How can I fix it?

System: Windows 7 x64. Mangoose is ideal for my project: I need xhr requests from local page (payment terminal use this plain html as templates).

Line:
extra_headers Access-Control-Allow-Origin: myotherdomain.com

Full config file:
# Mongoose web server configuration file.
# For detailed description of every option, visit
# https://github.com/cesanta/mongoose
# Lines starting with '#' and empty lines are ignored.
# To make a change, remove leading '#', modify option's value,
# save this file and then restart Mongoose.

# access_control_list 
# access_log_file 
# auth_domain mydomain.com
# cgi_interpreter 
# cgi_pattern **.cgi$|**.pl$|**.php$
# dav_auth_file 
# dav_root 
# debug 0
document_root D:\Projects\luna_php64\Samoplat-Paykiosk\kiosk\skins\
# enable_directory_listing yes
# error_log_file 
extra_headers Access-Control-Allow-Origin: myotherdomain.com
# extra_mime_types 
# global_auth_file 
# hide_files_patterns 
# hexdump_file 
# index_files index.html,index.htm,index.shtml,index.cgi,index.php
# listening_port 8080
# run_as_user 
# ssi_pattern **.shtml$|**.shtm$
# ssl_certificate 
# ssl_ca_certificate 
start_browser no
# url_rewrites

alashkin, 2016-11-15:
Hi! 
Try to run `mongoose <your config file name here>`

alashkin, 2016-11-21:
Closed due to inactivity

closed 2016-11-21 by alashkin

sibputnik, 2016-11-30:
@alashkin it not help.