cesanta/mongoose #732: websocket_message flag always has value 129, doesn't match with any of the opcodes

amitv87, 2016-11-01:
Here's my code. The server always sends text frame but the **wm->flags** never matches to **WEBSOCKET_OP_TEXT**.
Is there a way to check opcode of the received frame?
~~~
case MG_EV_WEBSOCKET_FRAME: {
  struct websocket_message *wm = (struct websocket_message *) ev_data;
  switch(wm->flags){
    case WEBSOCKET_OP_CONTINUE:{
      printf("WEBSOCKET_OP_CONTINUE");
      break;
    }
    case WEBSOCKET_OP_TEXT:{
      printf("WEBSOCKET_OP_TEXT");
      printf("%.*s\n", (int) wm->size, wm->data);

      const char* buff = "hello from pi";
      mg_send_websocket_frame(nc, WEBSOCKET_OP_TEXT, buff, strlen(buff));
      break;
    }
    case WEBSOCKET_OP_BINARY:{
      printf("WEBSOCKET_OP_BINARY");
      break;
    }
    case WEBSOCKET_OP_CLOSE:{
      printf("WEBSOCKET_OP_CLOSE");
      break;
    }
    case WEBSOCKET_OP_PING:{
      printf("WEBSOCKET_OP_PING");
      break;
    }
    case WEBSOCKET_OP_PONG:{
      printf("WEBSOCKET_OP_PONG");
      break;
    }
    default:{
      printf("unkown MG_EV_WEBSOCKET_FRAME code: %u\n", wm->flags);
      break; 
    }
  }
  break;
}
~~~

alashkin, 2016-11-15:
Hi!
>The server always sends text frame but the wm->flags never matches to WEBSOCKET_OP_TEXT

What values you get in `wm->flags`?
And another question: do you mean Mongoose-based server or some third party?
It is possible, what problem is on server side?

amitv87, 2016-11-15:
Here's the use case
**mongoose websocket client ---> uwebsockets sever**
and the server always sends text messages

I later resolved the issue by reading lower four bits of the flag as websocket opcode values range from **0x00 to 0x0F**
~~~
int opcode = wm->flags & 0x0F;
~~~
this now matches to the desired opcode :)

alashkin, 2016-11-15:
Yes, that's how documentation says (https://docs.cesanta.com/mongoose/master/#/c-api/http.h/mg_send_websocket_frame.md/). Not too clear though

closed 2016-11-15 by alashkin