cesanta/mongoose #734: NTLM Response not being sent as web socket handshake request

tfgsupport, 2016-11-09:
Hi,

I am using Mongoose to produce an NTLM challenge to a web browser when it tries to access the websocket. Below is the NTLM authorization header response generated by the browser but this is being received by Mongoose as a MG_EV_RECV (Data has been received) and not as a MG_EV_WEBSOCKET_HANDSHAKE_REQUEST which is what I was expecting. Why is this?

**Response:**
```

Host: 127.0.0.1:8000
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; rv:49.0) Gecko/20100101 Firefox/49.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-GB,en;q=0.5
Accept-Encoding: gzip, deflate
Sec-WebSocket-Version: 13
origin: http://127.0.0.1:8000
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: 6/doxO6QJWZScrXpIN4F+A==
Connection: keep-alive, Upgrade
Pragma: no-cache
Cache-Control: no-cache
Upgrade: websocket
Authorization: Negotiate TIRM***************************************************************************************
```

alashkin, 2016-11-10:
Hi,
You are building *server* based on Mongoose, right?
Does your server based on any examples shipped with Mongoose or it is written from scratch?
If the latter - didn't you miss `mg_set_protocol_http_websocket` call? 
Also, besides NTLM, does you server work or you also have another issues?

tfgsupport, 2016-11-10:
Hi,

Yes, based on Mongoose. It's written from scratch. I've checked and we haven't missed the mg_set_protocol_http_websocket call.

alashkin, 2016-11-10:
ok, let me try...

alashkin, 2016-11-10:
I tried to reproduce with Mongoose' example `simplest_web_server`  - compiled it, started and sent response from your first message to that server (via netcat).
And got `MG_EV_WEBSOCKET_HANDSHAKE_REQUEST` event.
I.e. it works.
I suggest you to discover  `simplest_web_server` example and find the differences with your code.
Also, you can create topic at http://forum.cesanta.com/ , paste your code there and probably someone will help.
Or, feel free to purchase commercial support, this is an option as well.

alashkin, 2016-11-10:
PS: just to be sure, you _can_ receive several MG_EV_RECV before MG_EV_WEBSOCKET_HANDSHAKE_REQUEST, it is normal. I.e. if you wait for first EV_RECV it is incorrect, wait for MG_EV_WEBSOCKET_HANDSHAKE_REQUEST

closed 2016-11-10 by alashkin