cesanta/mongoose #735: NTLM response sent in http_message uri field and not in http_message message field

tfgsupport, 2016-11-10:
Hi,

I am using Mongoose to produce an NTLM challenge to a web browser when it tries to access the websocket. The NTLM authorization header response generated by the browser is being received by Mongoose in the http_message uri field and not in the http_message message field, as below, which is what I was expecting. Why is this?

**Message field**

`0x0000025c <Error reading characters of string.>`

**URI field**

```
GET / HTTP/1.1
Host: 127.0.0.1:8000
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; rv:49.0) Gecko/20100101 Firefox/49.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-GB,en;q=0.5
Accept-Encoding: gzip, deflate
Sec-WebSocket-Version: 13
origin: http://127.0.0.1:8000
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: 7NfZKAHKacFjwgDxjmGBFQ==
Connection: keep-alive, Upgrade
Pragma: no-cache
Cache-Control: no-cache
Upgrade: websocket
Authorization: Negotiate TIRM********************************************************************
```

alashkin, 2016-11-10:
I'm not sure I'm understand
In #734 you've wrote, what you got MG_EV_RECV on this header.
But in this case, how you get http_message?

and..this is copy-paste from debugger, right? 
```
Message field

0x0000025c <Error reading characters of string.>
```

tfgsupport, 2016-11-10:
Hi Alashkin,

What can we expect for the event data in the case of MG_EV_RECV?

You're correct. I have a MG_EV_RECV on this header and got a http_message by doing the following

`struct http_message *hm = (struct http_message *) event_data;`

It seems the message happens to be in the uri field but I assume this is not always the case. Yes the above is copy-paste from the debugger.

alashkin, 2016-11-10:
In case of MG_EV_RECV event_data is a pointer to `int` with number of bytes received.

Thus, `struct http_message *hm = (struct http_message *) event_data;` is simple incorrect for MG_EV_RECV.
 
See here: https://docs.cesanta.com/mongoose/master/#/overview/events.md/

I'd suggest to close this issue and focus on #734

tfgsupport, 2016-11-10:
Agreed. The key thing for us is to work out why the NTLM response is not being received as a handshake request in which case we can use the http_message. You may close this issue.

closed 2016-11-10 by alashkin