cesanta/mongoose #736: time using double is very expensive in my embeded system

gghyoo, 2016-11-11:
In many embeded system, there are no fpu inside. The double float operation will cost much more time then integer operation， so you should change the double of time define to int64_t, I think this will help to reduce flash space and cpu time.

alashkin, 2016-11-14:
Hi!
What platform do you use? Is it really performance issue or you don't want to link with softfpu lib or something else?

gghyoo, 2016-11-21:
my cpu is mips using openwrt which don't have a fpu, float point computing is very slow.

alashkin, 2016-11-28:
But Mongoose doesn't use too expensive calculations... So, I'm not sure it's worth fixing. 
Maybe you have profiling results?

alashkin, 2016-12-06:
We are ready for discussion on our forum (http://forum.cesanta.com/)
Feel free to start topic. 
But at this moment, this change doesn't too useful.

closed 2016-12-06 by alashkin