cesanta/mongoose #737: use of mqtt broker over websocket and QOS

cordovapolymer, 2016-11-14:
Hi, mongoose is very promissing,
could it's mqtt broker be used over secure websocket? 
What QOS levels does the MQTT implementation support?

alashkin, 2016-11-14:
Hi!
At this moment, MQTT implementation doesn't support MQTT over WebSocket - only over TCP, but it supports SSL though 
As for QoS, can say it supports QoS 1, but keep in mind, that Mongoose is not MQTT just broker - it is Networking Library, thus, it can be expanded to support  QoS 0 and 2 as well.

mmikulicic, 2016-11-14:
@cordovapolymer, in which environment/platform do need a MQTT broker with WebSocket support ?

cordovapolymer, 2016-11-14:
I'd like to use it on Debian linux jessie/stretch as a broker for android clients which are written in javascript so its very convenient to use MQTT over WebSocket.

alashkin, 2016-11-15:
@cordovapolymer Can we close this issue? Or you have additional questions?

cpq, 2016-11-15:
@cordovapolymer could you elaborate on your project please? What is the use case? Is your project commercial/personal? Thanks!

cpq, 2017-08-24:
We have put together an example for mqtt over websocket server.
https://github.com/cesanta/mongoose/tree/dev/examples/mqtt_over_websocket_server
Only QOS0 is supported.

closed 2017-08-24 by cpq

cordovapolymer, 2017-11-26:
@cpq Do you plan to support QOS1&2?