cesanta/mongoose #739: mongoose binary leak

bjudd, 2016-11-17:
We have been running a simple webserver with Mongoose 5.6 binary as a service for displaying some information locally. The program in question refreshes data every 5 seconds, using xmlHTTPRequest POST. We found that each time, it creates and apparently never release 2 handle or thread counts. (monitoring software refers to them by different names). If there is no POST call, or no php file to handle it, there is no leak. Using the Mongoose 5.6 as an exe, and using the newer Mongoose 6.5 exe, it behaves the same. It has behaved the same on two different windows 10 machines (mine is a Windows 10 pro Version 10.0.14393 Build 14393, 64bit) and a 32bit Windows 7 machine. "GET" and "POST" behaved the same as far as leaks go.

Doing two xmlHTTPRequests every 5 seconds runs the PC out of memory in about a week. We get errors such as "An operation on a socket could not be performed because the system lacked sufficient buffer space or because a queue was full", and event logs such as "Activation of app Microsoft.CommsPhone_8wekyb3d8bbwe!App failed with error: Insufficient system resources exist to complete the requested service..."

The leak does not occur when running with IIS.

As my program is rather involved, I tried a simpler page from a sitepoint.com class. I modified the code so you could keep clicking the town names and calling the xmlHTTPRrequest. Each time it loads the "info", the count goes up by two. This can be viewed on Window's Performance Monitor or with Task Manager with a Handle column added, or a variety of other ways.

Any advice on making the code work with Mongoose, or making mongoose work with the code, would be appreciated. Mongoose is much easier to set up than IIS.

===========================================

altweatherwidget.js:

var WeatherWidget =
{
init: function()
{
var weatherWidget = document.getElementById("weatherWidget");
var anchors = weatherWidget.getElementsByTagName("a");

    for (var i = 0; i < anchors.length; i++)
    {
        $(anchors[i]).bind("click", WeatherWidget.clickListener);
    }
},

clickListener: function(event)
{
    try
    {
        var requester = new XMLHttpRequest();
    }
    catch (error)
    {
        try
        {
            var requester = new ActiveXObject("Microsoft.XMLHTTP");
        }
        catch (error)
        {
            var requester = null;
        }
    }

    if (requester != null)
    {
        var widgetLink = this;

        var timeoutHandler = function()
        {
            requester.abort();
            WeatherWidget.writeNoUpdate();
            WeatherWidget.writeError("The server timed out while making your request.");
        };
        widgetLink._timer = setTimeout(timeoutHandler, 100);

        var city = this.firstChild.nodeValue;

        requester.open("GET", "ajax_weather.php?city=" + encodeURIComponent(city), true);
        requester.onreadystatechange = function()
        {
            if (requester.readyState == 4)
            {
                clearTimeout(widgetLink._timer);

                if (requester.status == 200 || requester.status == 304)
                {

WeatherWidget.writeUpdate(requester.responseXML);
[... 227 lines cut ...]
===========================================

altweatherwidget.html:

<!DOCTYPE html>

Alt Weather Widget

ALTWeather Widget

Weather

Please select a city:

    London

    New York

    Melbourne

===================================================
ajax_weather.php:
<?php header("Content-type: text/xml"); $city = $_REQUEST["city"]; if ($city == "Melbourne") { $fileName = "melbourne.xml"; } else if ($city == "London") { $fileName = "london.xml"; } if ($city == "New York") { $fileName = "new_york.xml"; } $filePointer = fopen($fileName, "r"); $xmlData = fread($filePointer, filesize($fileName)); fclose($filePointer); print($xmlData); return true; ?>

sorry, i've not posted here before.  this is lifted from a discussion on forum.cesanta.com.  Sergey Lyubka agreed it was a bug and said to post here.

cpq, 2020-12-13:
Please use the latest code.

closed 2020-12-13 by cpq

bjudd, 2020-12-14:
in the intervening 4 years, we of course stopped using mongoose.  thanks anyhow.