cesanta/mongoose #740: websocket connection is interrupted and closed with code 1006 in binary messages with more than 1.8MB only in Chrome

lmiguelmh, 2016-11-18:
I needed to send some pdf files to a local websocket server and found that when sending a file with more than approx. 1.8MB the websocket "CLOSED" with code=1006 CLOSE_ABNORMAL (https://developer.mozilla.org/en-US/docs/Web/API/CloseEvent). This issue only happens with Chrome (54.0.2840) and not Firefox (47.0.1) and always in the same position (curiously enough this position varies depending on the file). The message never arrives to the server and the connection is closed.

I've added some log messages (I hope it is clear). I am using Chrome here:

	001F18C0 after fd=128 nc_flags=256 rmbl=1834831 smbl=0
	mg_if_poll MG_EV_POLL event
	mg_http_handler ??? event
	001F18C0 fd=128 fd_flags=1 nc_flags=256 rmbl=1834831 smbl=0
	001F18C0 8192 bytes (PLAIN) <- 128
	001F18C0 8192 1834831   <-----   here is the last position 1,834,831 bytes 
	mg_recv_common MG_EV_RECV event
	001F18C0 >BEF proto ev=3 ev_data=0045F070 flags=256 rmbl=1843023 smbl=0
	mg_ws_handler ??? event
	001F18C0 >BEF user ev=3 ev_data=0045F070 flags=256 rmbl=1843023 smbl=0
	MG_EV_RECV: 8192
	001F18C0 >AFT user flags=256 rmbl=1843023 smbl=0
	001F18C0 >AFT proto flags=1280 rmbl=1843009 smbl=0   <---   the flag changes?!

	001F18C0 after fd=128 nc_flags=1280 rmbl=1843009 smbl=0
	mg_if_poll MG_EV_POLL event
	mg_http_handler ??? event
	mg_close_conn MG_EV_CLOSE event   <-----   don't understand why the CLOSE event is generated
	001F18C0 1280 128
	001F18C0 >BEF proto ev=5 ev_data=00000000 flags=1280 rmbl=1843009 smbl=0
	mg_ws_handler ??? event
	001F18C0 >BEF user ev=5 ev_data=00000000 flags=1280 rmbl=1843009 smbl=0
	MG_EV_CLOSE
	127.0.0.1:3448 -- left

The same file doesn't have any problems in Firefox.

At first I thought there was some bug or problem in Chrome. I googled a little bit and didn't find anything... then I started to test with other websocket implementations and managed to send the file! (I used the Jetty javax.websocket example from https://github.com/jetty-project/embedded-jetty-websocket-examples with this small change http://stackoverflow.com/a/40684974/2692914). 

I used this sample https://github.com/cesanta/mongoose/tree/master/examples/websocket_chat
whit a small change in index.html to send a file trough a websocket:

    <div>
        <script type="text/javascript">
            old = onload;
            onload = function () {
                if (old)
                    old();

                var input = document.querySelector("#input-pdf");
                input.addEventListener('change', function (event) {
                    var fileReader = new FileReader();
                    fileReader.onload = function (e) {
                        ws.send(new Uint8Array(e.target.result));
                        //ws.send(e.target.result); //the same problem with ArrayBuffer
                    };
                    fileReader.readAsArrayBuffer(file);
                });
            };
        </script>
        <label>inpuFile</label>
        <input type="file" id="input-pdf" size="30"/>
    </div>

Maybe, maybe this problem is related? https://github.com/Gregwar/mongoose-cpp/issues/3
Thanks for this small and so powerful server!

lmiguelmh, 2016-11-24:
Okay, I've followed the issue to this function, _notice there are todo items_...
```
static int mg_deliver_websocket_data(struct mg_connection *nc) {
...
        *sizep = 0; /* TODO(lsm): fix. this can stomp over frame data */
...
      /* TODO(lsm): properly handle OOB control frames during defragmentation */
...
    /* If client closes, close too */
    if ((buf[0] & 0x0f) == WEBSOCKET_OP_CLOSE) {
		printf("(buf[0] & 0x0f) == WEBSOCKET_OP_CLOSE\n");
      nc->flags |= MG_F_SEND_AND_CLOSE;
    }
}
```
For some reason the function enters the if block even when there aren't any CLOSE messages.

alashkin, 2017-03-01:
Closed via https://github.com/cesanta/mongoose/pull/767

closed 2017-03-01 by alashkin