cesanta/mongoose #741: Sending a response to a MG_EV_HTTP_REQUEST waits a mg_mgr_poll interval

nkolban, 2016-11-21:
I'm using Mongoose embedded in Duktape (JavaScript on embedded) running in an ESP32.  In my Mongoose logic, I setup my environment and then run the polling loop:

```
while (1) {
   mg_mgr_poll(&mgr, 5000);
}
```

When an `MG_EV_HTTP_REQUEST` arrives in my event handler, my design doesn't allow me send a response immediately.  Instead I must add a message to a JavaScript queue and return from the event handler.  I save the `struct mg_connection`.

A very short time later the message is received from the queue which causes another C function to be called and it is in that C function that I want to complete the HTTP response.  In the C function I perform:

```
void myFunc() {
   mg_send_head(...);
   mg_send(...);
   g_nc->flags |= MG_F_SEND_AND_CLOSE;
}
```
where `g_nc` is the `mg_connection` saved from the `MG_EV_HTTP_REQUEST`.   What I am finding is that the logic all works ... however, the actual transmission and close doesn't seem to happen until after the `mg_mgr_poll()` timeout and restart.  If I were to hazzard a guess ... it is behaving as "though" the flags on the connection aren't being interrogated until the next call to `mg_mgr_poll()`.  That would make sense and it seems like my function should really be:
```
void myFunc() {
   mg_send_head(...);
   mg_send(...);
   g_nc->flags |= MG_F_SEND_AND_CLOSE;
   // Do something here to cause Mongoose to evaluate the connection flags
}
```

If I can add any more information, please let me know.

cpq, 2016-11-21:
I think you're pushing data to a connection from a different thread, which can potentially crash your program because mongoose API is not thread safe. All mongoose IO should be done in a single thread. Use `mg_broadcast()`:

```C
/*
 * Passes a message of a given length to all connections.
 *
 * Must be called from a thread that does NOT call `mg_mgr_poll()`.
 * Note that `mg_broadcast()` is the only function
 * that can be, and must be, called from a different (non-IO) thread.
 *
 * `func` callback function will be called by the IO thread for each
 * connection. When called, the event will be `MG_EV_POLL`, and a message will
 * be passed as the `ev_data` pointer. Maximum message size is capped
 * by `MG_CTL_MSG_MESSAGE_SIZE` which is set to 8192 bytes.
 */
void mg_broadcast(struct mg_mgr *, mg_event_handler_t func, void *, size_t);
```

Please refactor your code and make sure that the code that completes the reply, does NOT send any data to a connection, but rather sends data to the Mongoose thread via `mongoose_broadcast()`.

nkolban, 2016-11-21:
Thank you ... this makes great sense.  I'll rework my code exactly as you described.  Thank you for the super fast response.  Y'all are really on top of Mongoose and I'm loving using it.

closed 2016-11-21 by nkolban

cpq, 2016-11-21:
Thanks Neil!

You moved to ducktape, which is sorry to hear.
I believe we'll convince you to move back to our JS engine which we are refactoring at this moment, and which we aim to make extremely small, leaving plenty of RAM to the user code.

nkolban, 2016-11-21:
I'm drowning in JS engines at the moment ... Espruino, JerryScript, Duktape and the Cesanta JS engine.  The new ESP32 has 512K of RAM so the shift has moved from RAM space to ease of integration with a C environment.  However, when ready, I'll be delighted to assist with getting y'alls engine running on ESP32 in addition to the other platforms.  Ive got Espruino, JerryScript AND Duktape all running there just now.

cpq, 2016-11-21:
Very interesting!
Could you share which one do you like most and why?
Do you have any data on the footprint comparison?

nkolban, 2016-11-21:
Back in 2015 I stumbled across Espruino and my hobby at the time was ESP8266 so worked hard on porting  Espruino there.  However the amount of RAM on the ESP8266 was too low and we kept running out of RAM.  With the arrival of ESP32, I started to port Espurino there and got it working ... however the battle was all up-hill.  The code base was muddled and complex.  Someone then pointed me to JerryScript and it short order, I got it running on ESP32.  However because I need good C bindings, I found it rather cumbersome to keep interacting with JerryScript in C by creating JS types from C types and then passing those around and then bringing them back to C types.  It was then that I came across Duktape ... and it compiled for ESP32 first time and ran straight away.  It has (opinion) better C language interfaces.  As for the Cesanta JS ... I'm afraid I haven't tried that yet.  As of right now, I have what I need and don't yet have a compelling reason to change (that I am aware off).  The secondary considerations will likely come when I look at resource utilization and performance ... but I haven't gone there yet.

cpq, 2016-11-21:
Thanks Neil.
We also found out that writing C/JS glue code is a hard task. We plan to have an API to import native C functions into the JS similarly to ruby's Fiddle, http://ruby-doc.org/stdlib-2.0.0/libdoc/fiddle/rdoc/Fiddle.html . Basically, we plan to have a JS api `import(c_function_name, return_type, params)` which would able to make a native C function visible to the JS. No glue code would be necessary.

Note that our engine is already so small that it can do tasks no other engines can. For example, http://mongoose.cloud/downloads/aws/README.txt - a working example on how to control your ESP8266 remotely using AWS IoT. This example can talk to the AWS IoT via two-way SSL (this task is on itself is accomplished by few), but I can get JS snippets over MQTT, execute them and send replies - so a fully working server with SSL and JS, on ESP8266.

We plan to make our JS even smaller, make it working on 8- and 16-bit micros. But that is another topic.