cesanta/mongoose #744: mg_broadcast and intra-process communication using sockets ...

nkolban, 2016-11-21:
I am working on the port of Mongoose to ESP32 and hadn't previously exercised `mg_broadcast`.  When I tried, I found it would not work.  After a solid day of debugging, I have narrowed it down to the use on a socket pair at the datagram level.  Apparently the logic is to create a pair of sockets, connect them to each other and then the Mongoose event handling / polling thread includes one end in the wakeup set.  When an mg_broadcast is performed, we write to the other end of the socket.  What I am finding (on ESP32) is that the sending end works but the receiving end doesn't receive the datagram.  I have validated this in pure C/ESP32 code and created an issue against the ESP32 software ... see:

https://github.com/espressif/esp-idf/issues/114

The purpose of this issue is to cross link the two.  If ESP32 has an issue with this usage pattern of sockets, then it may need an alternate design for Mongoose running on ESP32.  I also raise this issue here in case a reader who has worked in this area previously might have a memberance or an aha moment to add some light.

cpq, 2016-11-22:
Socket pair is using localhost loopback `127.0.0.1` to create a connected pair of sockets.

At least on ESP8266, localhost loopback was disabled in LWIP. Maybe the same is the case for ESP32 - please check this out, and switch it on.

nkolban, 2016-11-22:
Thank you guys ... looking here ... http://esp32.com/viewtopic.php?f=13&t=544 ... it appears that there is a known bug.

Thinking ahead ... if using a loopback were not possible ... does that mean the mg_broadcast() would simply not function and hence I would not be able to invoke mg_function() from a different thread?

cpq, 2016-11-22:
Let us think about the workaround ..

cpq, 2016-11-24:
Happy Thanksgiving Neil!

Could you wait for our own port of mongoose iot to esp32 please?
We anticipate it to be finished somewhere around the end of next week, or the week after next. `mg_broadcast` will be implemented using different mechanism (rtos queues).

nkolban, 2016-11-24:
Awesome ... and please let me know when available.  One of the things I am trying to do is write a book on ESP32 ... see https://leanpub.com/kolban-ESP32 ... and I'd love to cover Mongoose in detail.  So with your permission, I'd like to cover that too.

cpq, 2016-11-25:
Neil,
I've enjoyed your book on esp8266. It's great you're writing one on esp32!
I'd be delighted if you cover mongoose!

Also, as I said, we're working on refactoring our JS engine. There is something cool that is getting out of it. The byte code is actually a tiny Forth-like code, a low-level stack machine. JS parser compiles a subset of ECMA6 JS into that forth VM. That VM is really tiny, we think it might be possible to run it on 8-bit MCUs. And it would be easy to make a backend for other languages (e.g. ruby, python or java) and utilize the same Forth VM.
You'll have a chance to compare the footprint of different VMs. You're the only person I am aware of, who managed to collect such a zoo of embedded VMs. That gives a good position to compare on footprint, performance, JS/C glue.

alashkin, 2016-11-29:
@cpq 

Shall we close this issue? There is nothing to do here _for mongoose_.

cpq, 2016-11-29:
Yes, let's close it.
Neil, I'll ping you personally when we have ESP32 port ready to try.
Thank you!

closed 2016-11-29 by cpq

nkolban, 2016-11-29:
Thanks guys ... I hadn't closed it myself since I was tracking it for Mongoose/ESP32 availability ... but based on your offer to ping me when Mongoose on ESP32 is ported, that's perfect.  The mongoose component is vital in this sample that we hope will be useful to ESP32 folks:

https://github.com/nkolban/esp32-snippets/tree/master/networking/bootwifi

cpq, 2016-12-28:
@nkolban , ESP32 is now available!
You can start with https://github.com/cesanta/mongoose-iot/tree/master/fw/examples/c_mqtt

nkolban, 2016-12-28:
Awesome ... will get to it ASAP.  Thank you for the job guys ... can't wait to start playing.

cpq, 2016-12-28:
@nkolban thank you for giving it a try and making a note on ESP32 forum, much appreciated !